preesm/preesm #24: Initialization of delay tokens

jheulot, 2015-02-12:
It would be nice to have the possibility to initialize delay token by a configure actor or interface. It will simplify many feedback looped graphs.

antoine-morvan, 2017-10-27:
See #45

closed 2017-10-27 by antoine-morvan

Martayugui, 2019-08-19:
@antoine-morvan, given that there is no implementation yet (#45), is there any work-around to perform the initialization of a delay? I am trying to implement iterative loops and I cannot find the way.

antoine-morvan, 2019-08-20:
There is support for delay initialization since version 2.14.0 https://github.com/preesm/preesm/releases/tag/v2.14.0
Theory is described in this paper : https://hal.inria.fr/hal-01850252
The graph editor still has some issues handling this kind of setup (see https://github.com/preesm/preesm/issues/83 for instance).