preesm/preesm #28: [Bundled release] Run as menu and workflow task properties instability

kdesnos, 2015-08-17:
In the bundled release of preesm generated by the not-yet-public jenkins server:
- "Runs as" option in the right-click menu disappear after its first use.
- Properties of the workflow tasks accessed through the workflow editor sometimes also disappear.
- *.pi files are not visible in the package explorer.

antoine-morvan, 2017-05-12:
Could you confirm this is still valid with new product releases ?
See http://preesm.sourceforge.net/gensite/products/latest/

kdesnos, 2017-05-17:
It seems OK.

antoine-morvan, 2017-06-01:
It's actually still the case when installing from update site with latest version.

antoine-morvan, 2017-06-06:
There is now a new menu entry in the Preesm context menu to run workflows. Also the context menu within the graphical editor now allows to run a workflow.

closed 2017-06-06 by antoine-morvan