preesm/preesm #31: [Workflow] Can't open workflow files

skanur, 2017-05-07:
I'm running PREESM from source under develop branch (commit f6076bae4f6efb1b4a2b4a862337e2603ae09b6a). When I try to open a workflow associated with any project using default viewer, I get error message in the palette view. I have installed the pre-requisites and executed the source code as mentioned [here](http://preesm.sourceforge.net/website/index.php?id=building-preesm). 

Apologies for attaching error as pictures. I had no other option here as it appears in palette view with no option to select the text and I can't type the entirety of error message here.

[image]
[image]

antoine-morvan, 2017-05-07:
Hi, 

I was able to reproduce. The issue comes from a recent renaming of the plugins in the Graphiti project. Some ID constants in the code were not updated correctly after that renaming. This causes the configuration loader to miss part of the configuration, hence the exception.

I just fixed that on a Graphiti branch. I will merge and deploy everything next week. You will have to reinstall the Preesm dev meta feature (I'll bump this thread when it's done).

Until then, the **workflow runner is still working**. You can still right click and execute the workflow.

If you need to edit the workflow before that, you can either use older version of the editor (available as eclipse plugins on the update site here http://preesm.sourceforge.net/eclipse/update-site-backups/complete-2.2.5/), or edit using a text editor (workflow files are XML-like).

Thanks for the report.

skanur, 2017-05-08:
Although I can't seem to open Workflow editor in specified (show nothing but a blank), I'm using an xml editor for now. I can wait for the fix.

Thanks for the update.

skanur, 2017-05-08:
Update:

Looks like this issue affects other XML files as well. I get the same error when I try to see graphml files as well. Due to this I can't see the output SDF graphs (flatten & HSDF output). Reading XML of these files are tedious for a complicated graph output.

So could you please push the update as soon as you can?

Thanks a lot!

antoine-morvan, 2017-05-08:
Hi, 

I just push the update sites. I tried with a fresh install and all editors should work (only tried with the sobel example from tutorial). You could also update your current eclipse (Help > Check for Updates).

Let me know if you encounter any issues.

skanur, 2017-05-09:
Everything works fine now. Thank you!

closed 2017-05-09 by skanur