preesm/preesm #32: Cannot register new workflow tasks

skanur, 2017-05-10:
Section 4 of creating a workflow [tutorial](http://preesm.sourceforge.net/website/index.php?id=new-workflow-task) describes registering new workflow tasks. In particular Step 5 and 6 cannot be performed for extensions _org.ietr.dftools.workflow.*_ i.e. for all the worflow tasks provided by PREESM, right click does not show input and output. 

Infact, right clicking on extensions in Manifest.MF of plugins offered by PREESM by default (say org.ietr.preesm.algorithm) doesn't show input and output. However, I get this menu when I repeat the procedure 1 until 4 for plugins provided by eclipse and not PREESM. This is critical as workflows can't be registered and run.

Using Eclipse Modeling Tools Version: Neon.3 Release (4.6.3) Build id: 20170314-1500. At commit 385fc93b. 

Also in Step 1 of the page, nothing is mentioned if _Activator.java_ needs to be generated as well. Shouldn't we tick the box to generate them (its not selected by default)?

antoine-morvan, 2017-05-10:
Hi, 

I'll try to reproduce the issue.

As for Activator.java, it can be let unchecked. Activators are called upon plugin activation during Eclipse execution. They are usually used to initialize non static constants/configuration/preferences. I'll add a word about it.

antoine-morvan, 2017-05-10:
The issue is caused by missing .exsd schemas in the binaries of DFTools. The file build.properties in project org.ietr.dftools.workflow is missing the folder /schema/. Thus it's working when you have the source of DFTools in the workspace, and not when you use the new setup with Eclipse plugins from update site.

Similar issue exists in org.ietr.preesm.codegen.xtend plugin in Preesm. 

Including these schemas in the binaries fixes the issue. I will deploy a new version asap.

skanur, 2017-05-10:
Thanks a lot for the quick response and fix! It works flawlessly now.

closed 2017-05-10 by skanur