preesm/preesm #33: [Graph Transfo] Same parameter names in different unflattened hierarchical actors not supported

julienhascoet, 2017-05-12:
The graph transformation fails when the single rate validates the model onto multiple unflattened hierarchical actors which have a same parameter name (in different hierarchical actors).

antoine-morvan, 2017-05-12:
After short discussion:
- First step would be to integrate a pass that checks the SDF for validity (and throw meaningful error messages);
- Second, have a pass that rename actors when SDF is not valid;
- Finally, have the algorithms actually handle identical names in sub/parent graph parameters.

antoine-morvan, 2017-06-23:
Could reproduce on the small attached example [issue33.zip](https://github.com/preesm/preesm/files/1097969/issue33.zip)

Several hints from the cause:
*   The feeding of JEP may not take care of the graph hierarchy scoping (some sub graph parameter values are overridden by top graph parameter values, if parameters have same name);
*   Some errors are not explicit at all (with wrong conditions, such as "could not flatten because rate does not match: 4 != 4"). Some confusion when using hierarchical actor's production rate instead of subgraph I/O interface rates could be the cause.

closed 2017-06-26 by antoine-morvan