preesm/preesm #37: Workflow fails when graphs have special actors

skanur, 2017-05-18:
When graphs have special actors such as _Broadcast_, PREESM throws an runtime exception and dies. 
The workflow works fine if I remove all the special actors. The issue can be replicated with examples of stereo vision.
The error is given below:
```
eclipse.buildId=4.6.3.M20170301-0400
java.version=1.8.0_121
java.vendor=Oracle Corporation
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_GB
Framework arguments:  -product org.eclipse.platform.ide
Command-line arguments:  -product org.eclipse.platform.ide -data /home/sudeep/eclipse_workspaces/preesm/../runtime-EclipseApplication -dev file:/home/sudeep/eclipse_workspaces/preesm/.metadata/.plugins/org.eclipse.pde.core/Eclipse Application/dev.properties -os linux -ws gtk -arch x86_64 -consoleLog

org.eclipse.core.jobs
Error
Thu May 18 18:07:27 EEST 2017
An internal error occurred during: "Launching _org.ietr.preesm.stereo_Workflows_CodegenMemoryScriptsMixedMerged.workflow".

java.lang.NoSuchMethodError: org.eclipse.xtend2.lib.StringConcatenation.append(Ljava/lang/String;)V
	at org.ietr.dftools.algorithm.model.sdf.transformations.SpecialActorPortsIndexer.addIndexes(SpecialActorPortsIndexer.java:176)
	at org.ietr.preesm.pimm.algorithm.pimm2sdf.visitor.StaticPiMM2SDFVisitor.visitPiGraph(StaticPiMM2SDFVisitor.java:101)
	at org.ietr.preesm.experiment.model.pimm.impl.PiGraphImpl.accept(PiGraphImpl.java:414)
	at org.ietr.preesm.experiment.model.pimm.visitor.impl.PiMMVisitorImpl.visit(PiMMVisitorImpl.java:74)
	at org.ietr.preesm.pimm.algorithm.pimm2sdf.visitor.StaticPiMM2SDFVisitor.visitPiGraph(StaticPiMM2SDFVisitor.java:119)
	at org.ietr.preesm.experiment.model.pimm.impl.PiGraphImpl.accept(PiGraphImpl.java:414)
	at org.ietr.preesm.pimm.algorithm.pimm2sdf.visitor.StaticPiMM2SDFVisitor.visitPiGraph(StaticPiMM2SDFVisitor.java:92)
	at org.ietr.preesm.experiment.model.pimm.impl.PiGraphImpl.accept(PiGraphImpl.java:414)
	at org.ietr.preesm.pimm.algorithm.pimm2sdf.StaticPiMM2SDFLauncher.launch(StaticPiMM2SDFLauncher.java:91)
	at org.ietr.preesm.pimm.algorithm.pimm2sdf.StaticPiMM2SDFTask.execute(StaticPiMM2SDFTask.java:75)
	at org.ietr.dftools.workflow.AbstractWorkflowExecutor.execute(AbstractWorkflowExecutor.java:336)
	at org.ietr.dftools.ui.workflow.launch.WorkflowLaunchConfigurationDelegate.launch(WorkflowLaunchConfigurationDelegate.java:100)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:885)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:739)
	at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1039)
	at org.eclipse.debug.internal.ui.DebugUIPlugin$8.run(DebugUIPlugin.java:1256)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
```

On some googling, looks like org.ietr.dftools.algorithm links with xtend 2.10 or earlier.

antoine-morvan, 2017-05-19:
Hi, 

I had a similar (offline) report with Preesm 2.2.8: the dev meta feature was including its requirements instead of importing them (see links below for details). The inclusion states that all dependencies are packaged with the feature, whereas the import states that dependencies needs to be installed first. In version 2.2.8 (and earlier), the dev feature was including all its plug-in dependencies. Thus it would force some versions for some plug-ins, and can cause conflicts, as it is in your case with xtend and xbase version conflict (xtend 2.11 uses a new StringConcatenation API from xbase that was not present in version 2.10).

For version 2.2.9 of Preesm, I changed all dependencies from plug-in inclusions to feature imports so that the version is not hard coded in the feature. This also prevents version conflicts with currently installed features. In order to fetch missing features and/or help resolve update issues, I also added few discovery sites (see 'Sites to Visit' in [Eclipse documentation](http://help.eclipse.org/kepler/index.jsp?topic=%2Forg.eclipse.pde.doc.user%2Fguide%2Ftools%2Feditors%2Ffeature_editor%2Finformation.htm&cp=4_3_2_1_1)).

For the record, could you report the Eclipse installation details of your Eclipse distribution for Xbase and Xtend  (Help > About Eclipse > Installation Details > Plug-ins then filter xbase then xtend) ? (example [here](http://imgur.com/a/1GOxB))

To solve the issue, try to Help > Check for Updates. Updating to latest Preesm should update the dev feature, and all xtend/xbase versions should be updated to 2.11.

*   http://stackoverflow.com/questions/22966936/feature-xml-requires-vs-includes
*   [Included Features](http://help.eclipse.org/kepler/index.jsp?topic=%2Forg.eclipse.pde.doc.user%2Fguide%2Ftools%2Feditors%2Ffeature_editor%2Fincluded_features.htm&cp=4_3_2_1_3)
*   [Feature Dependencies](http://help.eclipse.org/kepler/index.jsp?topic=%2Forg.eclipse.pde.doc.user%2Fguide%2Ftools%2Feditors%2Ffeature_editor%2Fdependencies.htm&cp=4_3_2_1_4)

skanur, 2017-05-19:
Hi,

Thanks for the detailed response. The issue was with 2.10 versions of xtend and xbase installed by default in eclipse EMF tools. I had to add update sites from [here](https://www.eclipse.org/Xtext/download.html) and update Xtend (only) to 2.11.

closed 2017-05-19 by skanur