preesm/preesm #38: How is this graph schedulable?

skanur, 2017-09-15:
Hello!

The "schedule check" of SDFGraph class reports below graph as schedulable. However, I don't believe this should be the case. 
[image]

The port rates of actor:
Z: (outZ, 1)
A: (inA, 1) (outA, 2) (inBA, 2) (inCA, 1)
B: (inB, 3) (outB, 3) (outBA, 3)
C: (inC, 2) (outC, 1) (outCA, 1)
D: (inD, 1)

Delay tokens between actor A and B is 6 and delay tokens between **actor A and C is 1**.

My local copy of PREESM is not updated yet, that could be a reason?

kdesnos, 2017-09-15:
> unschedulable_network

Hi Sudeep,

Yes indeed! I think the explanation is that your graph is "consistent" 
but not schedulable.

The graph is consistent because, provided that there are enough delays 
on the cycles, it will be possible to find a fixed sequence of actor 
firings that, when executed, brings back the FIFOs to their initial 
state. If I'm not mistaken, the repetition vector should be 3Z, 3A, 2B, 
3C, 3D.

The graph however is not "schedulable" because there are not enough 
tokens in AC to complete a graph iteration.

If you use a complete workflow in Preesm, this issue will be detected, 
at the latest, when generating the DAG during the Mapping.

As a conclusion, there are two options there. Either rename the 
"schedulability check" into "consistency check", OR complement its code 
to verify both consistency and schedulability.

Can you please fill a bug report on the git repository? (or correct the 
code yourself if you feel more adventurous ;).

Best,

Karol

skanur, 2017-09-16:
Hey Karol!

Thanks for the explanation! 

So if I understand, then there is no check to assure deadlock freeness until the creation of DAG. 

I will create an issue and try to implment it when I can :)

closed 2017-09-16 by skanur