preesm/preesm #40: Problems with the tutorial "Automated Measurement of Actor Execution Time"

Alberto18, 2017-09-22:
[org_ietr_preesm_sobel.zip](https://github.com/preesm/preesm/files/1324879/org_ietr_preesm_sobel.zip)
Hello, I would like to ask you for help with the tutorial of Measurement. I would like to apply it to a video stabilizator using linux. After following the instructions given in the PREESM webpage, I can´t read any data or number in the "Excel" document created, which is called Analysis.  
Thanks for your attention.
@mpelcat @kdesnos

mpelcat, 2017-10-04:
​​​​​Hi Alberto,

I tested with Preesm 2.5.0 on Linux64 and GCC 4.8.4. 

On the 1 core scenario and the timing tutorial, there was a problem with c99, I added to the CMakeLists.txt (see enclosed file):
set_property(TARGET tgt PROPERTY C_STANDARD 99)

I still get C errors but they seem to be in your code, am I wrong? In particular, there is a duplicated function computeBlockMotionVector functions (in computeBlockMotionVector.c and stabilization.c). Once you get the code corrected, compiled and run, it should fill the analysis.csv file when run.

Best Regards,
Maxime

CMakeFiles/sobel.dir/src/stabilization.c.o: In function `computeBlockMotionVector':
stabilization.c:(.text+0xe70): multiple definition of `computeBlockMotionVector'
CMakeFiles/sobel.dir/src/computeBlockMotionVector.c.o:computeBlockMotionVector.c:(.text+0x0): first defined here
CMakeFiles/sobel.dir/generated/main.c.o: In function `main':
main.c:(.text.startup+0x0): multiple definition of `main'
CMakeFiles/sobel.dir/src/main.c.o:main.c:(.text.startup+0x0): first defined here
CMakeFiles/sobel.dir/generated/main.c.o: In function `main':
main.c:(.text.startup+0x2c): undefined reference to `communicationInit'
collect2: error: ld returned 1 exit status

[CMakeLists.txt](https://github.com/preesm/preesm/files/1356162/CMakeLists.txt)

Alberto18, 2017-10-04:
Hi again Maxime, 

first of all, thanks for your attention. I have already added the line you wrote to the CmakeList.txt.
Despite the fact that, as you have said, I have the function duplicated (because I use the stabilization function in the "1 core" scenario and the "computeBlockMotionVector" function to parallelize in the "4 core" scenario), I don't get any error warning as you can see in the photo below. Moreover, the programme creates the "Solution Gantt"! Despite this, can that duplicated function be the problem of still not working?

After running the Instrumented.codegen, I only run the video processing from the command window. And after finishing, I just check the Analysis archive. Is that method the correct one? Should I do anything else?

Thanks for your attention again.
Best Regards.
Alberto

[image]

mpelcat, 2017-10-05:
Hi Alberto,
It seems to me you are missing one step. Launching the workflow, as shown in the screenshot, only generates the C code for your project. The Gantt chart is generated from the numbers stated in the scenario but does not mean the code is being executed. After that, you need to compile and launch your C project to obtain the real timings (and feed them back in the scenario). I had the previously evoked compilation problems in GCC, not in the Preesm GUI itself. For the compilation procedure, you can refer to tutorial 2. It uses Cmake.
Best,
Maxime

antoine-morvan, 2017-11-13:
Marking as invalid. Please reopen if you can provide more data regarding an issue in the tool or in the documentation.

closed 2017-11-13 by antoine-morvan