preesm/preesm #56: Generate header file containing all parameters definitions

ahonorat, 2018-03-19:
In the preesm static version, it would easier applications development if all parameters defined in preesm could be exported to a header file during the code generation. It avoids adding dependencies between parameters and actors when they are not impacting the port rates but impacting the actor internal working (its *refinement*).
I propose to generate all parameters of all hierarchy levels in a single header file, using the C convention (names are capitalized, special characters replaced by underscores), with an option to specify the path to the generated header.

antoine-morvan, 2018-03-19:
Parameter names in the generated header should be also prefixed (or suffixed) so as to not overlap with variable names having the same identifier. I.e. the following example would produce compilation error:

'''
// in params.h
#define MYPARAM 55

// in actor.h
#include "params.h"
void myActor(int MYPARAM /* ... */) {
'''

ahonorat, 2020-02-16:
Done with 9c35afa

closed 2020-02-16 by ahonorat