preesm/preesm #58: group deletion

orlyn44, 2018-03-20:
Doesn't do anything when trying to remove a mixed group of parameters and actors.
When removing a group composed only of actors, it works fine. Same for parameters.
The issue raises in the specific case of trying to remove parameters and actors in the same selected group.

orlyn44, 2018-03-20:
The problem is a bit more specific than expected. 
The bug seems to happen when a delay is in the selection group.

orlyn44, 2018-03-20:
[delay.pi.zip](https://github.com/preesm/preesm/files/1828614/delay.pi.zip)

antoine-morvan, 2018-09-03:
I could not reproduce the issue with the provided example. Has it been solved with more recent versions ?

antoine-morvan, 2018-10-22:
@orlyn44 please reopen if you can provide an example to reproduce with latest versions.

closed 2018-10-22 by antoine-morvan