preesm/preesm #61: Null Pointer when Code generation directory is invalid

ZhangZhijing1, 2018-06-13:
When the specified code generation directory specified in the scenario is invalid (i.e. it is not a path to an existing directory in the current workspace), the following NullPointerException occurs during the execution of the code generation workflow task:

> 
> 
> 09:37:24 Unexpected Exception: java.lang.NullPointerException
> 	at org.ietr.preesm.codegen.xtend.task.CodegenEngine.preprocessPrinters(CodegenEngine.java:249)
> 	at org.ietr.preesm.codegen.xtend.task.CodegenTask.execute(CodegenTask.java:104)
> 	at org.ietr.dftools.workflow.AbstractWorkflowExecutor.execute(AbstractWorkflowExecutor.java:355)
> 	at org.ietr.dftools.ui.workflow.launch.WorkflowLaunchConfigurationDelegate.launch(WorkflowLaunchConfigurationDelegate.java:88)
> 	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:885)
> 	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:739)
> 	at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1039)
> 	at org.eclipse.debug.internal.ui.DebugUIPlugin$8.run(DebugUIPlugin.java:1256)
> 	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
> 
>  Contact Preesm developers if you cannot solve the problem.

A more informative message would be more helpful (and less troubling) for the developer.

antoine-morvan, 2018-06-13:
Hi, 

I could not reproduce this issue when simply putting a random string as codegen path in the scenario. Could you provide a minimal example (Preesm project) to reproduce ?

ZhangZhijing1, 2018-06-13:
Hi,
For instance, when I duplicate a preesm project, the path of the Code Generation Directory in the new scenario remains the same as the path in the original project. If I don't change this directory, a Null Pointer Exception will be given in the console.

Thank you

antoine-morvan, 2018-06-13:
Could reproduce after deleting said copied project.

antoine-morvan, 2018-06-13:
Merged in develop.

commit 3b888279 (2018-06-13): (UI) Fix #61
commit 466045ad (2018-06-13): (UI) Fix #61
commit 64054650 (2018-06-13): (UI) Fix #61
closed 2018-06-13 by antoine-morvan