preesm/preesm #63: Configure output port behave awkwardly

jheulot, 2018-07-19:
I succeed to achieve this graph with Preesm PiSDF editor. 
It should not be possible to have a dependency and a fifo connected to the same port. 
[image]

Moreover, is it possible to have a config output port connected to a data port ?
[image]

[top.pi.txt](https://github.com/preesm/preesm/files/2209070/top.pi.txt)

antoine-morvan, 2018-07-19:
https://github.com/preesm/preesm/blob/5051bef2ff5534a427a69cd0245b87e4f631b1fe/plugins/org.ietr.preesm.ui.pimm/src/org/ietr/preesm/ui/pimm/features/CreateFifoFeature.java#L273-L278

Seems expected. Let's check what the semantic should be....

antoine-morvan, 2018-08-29:
Actually, keeping this behavior would:
- allow multiple edges to be connected to one port. That is in contradiction with all the other port behavior;
- allow a config port to be connected to a Fifo. That is in contradiction with ConfigInputPort behavior;

Thus, ConfigOutputPort connections will be restricted to dependencies only.

antoine-morvan, 2018-08-29:
Well, this can actually lead to complex constructions when forwarding parameter values several level up in the hierarchy... Moreover, the Spider support is not clear. Let's postpone resolution of this issue.