preesm/preesm #65: (Parameters) Evaluation of expressions cast to integer

orlyn44, 2018-07-26:
Parameters expression are evaluated in float and then cast to integer which can result in unconsistent behavior.

Example: ParameterA = 3/2 -> 2
                ParameterB = 2 * (3/2) -> 3 and not 4 as expected from the result of ParameterA.

Solution would be to display a warning in case of ParameterA when the cast induce either a ceil or a floor.
"Warning: Expression of parameter ParameterA was casted to integer resulting in loss of precision. To suppress this warning, use ceil/floor operators".

antoine-morvan, 2018-07-26:
Test : compute the delta from the "rounded" value and the original value, and display a warning only of delta is greater than some epsilon (say 0.0001).

hmiomand, 2025-01-25:
As of 5e43555e9e1ec072be907ff5c35cfd2bca44bd21, Parameter expressions are evaluated as double and generate a warning when a non-integer resolution is used for fifo rates.

closed 2025-01-25 by hmiomand