preesm/preesm #83: Autolayout with getter actor on delay

orlyn44, 2018-09-27:
There is a bug of the autolayout a getter actor is connected to a delay causing the 
graph editor to freeze.

It seems that the autolayout falls in an infinite while loop in that case.

Martayugui, 2019-08-23:
@antoine-morvan , I am in the same situation and I cannot debug my application because my autolayout is deleted once .diagram is generated. Is there any plan to fix this issue or do you know any work-around?

Thanks in advance,

kdesnos, 2019-08-26:
To whoever is going to try fixing this issue:
- The most urgent matter is to make sure that the editor does not freeze/crash in such a scenario. 
- Then, as a secondary (and less urgent) matter, it is to make sure the auto-layout produces a nicely organized graph.

commit 4703e05b (2019-09-04): Fix #74, Fix #83, Fix #128, Fix #190
commit 6ea6007f (2019-09-18): Fix #74, Fix #83, Fix #128, Fix #190