preesm/preesm #86: Properly handle multi-refinements and accelerators

antoine-morvan, 2018-10-22:
Currently, it is not possible to
* model an accelerator in the SLAM MoA;
* specify multiple implementations (or refinement) for the same actor with the scheduler knowing the specific timings of a given implementation on a given processing element;

Discussions should be conducted to define how the SLAM, PiSDF, scenario, and codegen have to be impacted in order to properly handle this.

Possible situations
* mono- and multi-core host
* synchronous and asynchronous accelerator
* shared accelerator(s): one accelerator can be triggered from several "master" cores
* non-slave accelerators (equivalent to DMA ?)
* reconfigurable hardware

### Use Cases

* Superscalar high-end homogeneous desktop multicore CPU(s) with GPGPU(s) running
  * CUDA
  * opencl
  * vulkan
  * does the API actually matter ?
  * i.e. i7 with 2*Titan running CUDA
* embedded system dev board with heterogeneous arm and GPGPU
  * different from above ?
  * i.e. Odroid XU3 with Exinos 5 and Mali GPU
* Dedicated hardware
  * consider programing time
  * how to deal with reconfiguration ?

leos313, 2018-10-22:
- how to deal with reconfiguration? ---> when we use PREESM (all the parameters known at design time), do we really need a reconfiguration? Maybe this can be an issue for SPiDER, in a second step. For now can be enough to have a fixed number of accelerators that doesn't change: they are there like a CPU is always there in the static case
- shared accelerator(s): one accelerator can be triggered from several "master" cores ---> this is already possible when we speak about hardware accelerators and some example are available.

antoine-morvan, 2018-10-23:
@leos313 I don't completely agree. If the scheduler finds out that re-configuring the hardware to execute another actor every time, then even with a static schedule we would have to deal with reconfiguration.

leos313, 2018-10-23:
@antoine-morvan In the case you were highlighted, I completely agree with you.
To solve it: Does PREESM take into account the "init" functions for the refinements, when they are present? If so, the reconfiguration can be performed there and, consequentially, be scheduled there (just before the actor execution).

orlyn44, 2018-10-23:
@leos313 The init functions are only called on start of the execution of the application, not at every start of a graph iteration. So, it would not be possible to use them for reconfiguring your HW like that. However you could add an actor in your graph that does just that.

leos313, 2018-10-24:
@orlyn44 Now it works exactly like that. However, in this case, an actor of the PiSDF is changing the MoA (or parameters strictly related to it).