preesm/preesm #89: stdint visual studio

orlyn44, 2018-12-17:
On auto generation of preesm files in windows there are some error for visual 2015.

in preesm_gen.h and dump.h this should be changed:

#include <stdint.h> 

#ifdef _MSC_VER
typedef _int32 int32_t;
typedef unsigned _int32 uint32_t;
#ifdef int64_t
#undef int64_t
typedef _int64 int64_t;
#endif 
#ifdef uint64_t
#undef uint64_t
typedef  unsigned _int64 uint64_t;
#endif

#else 
#include <stdint.h>
#endif

antoine-morvan, 2018-12-17:
Fixed in develop.

orlyn44, 2018-12-18:
Codegen for visual still not working properly.
The following
```
#ifdef int64_t
#undef int64_t
typedef _int64 int64_t;
#endif 
#ifdef uint64_t
#undef uint64_t
typedef  unsigned _int64 uint64_t;
#endif
```
should be changed to 
```
#ifdef int64_t
#undef int64_t
#endif 
typedef _int64 int64_t;
#ifdef uint64_t
#undef uint64_t
#endif
typedef  unsigned _int64 uint64_t;
```

antoine-morvan, 2018-12-18:
Fixed on develop.

commit 6eba0663 (2018-12-18): (Codegen) Fix #89
closed 2018-12-18 by antoine-morvan