preesm/preesm #91: (Gui) Out of sync

kdesnos, 2018-12-17:
When editing a graph, error tend to occur, causing the pi to be out of sync with the diagram, and the diagram to be not "saveable".

kdesnos, 2018-12-17:
To reproduce, follow tutorial on sobel and toy around by adding and removing actors and fifo without saving the graph in the middle. This should do the trick.

LowellHuawei, 2019-04-02:
Yes, I think my problem is related to this.

Sometimes detecting and fixing problems can be very difficult.  One possible way to alleviate the pain of this kind of problem, where 2 files get out of sync, is to make the save code write to temporary files and then only if both are successfully saved, delete the originals and rename the new saved copy.  It doesn't fix the problem, but it alleviates the fairly fatal error of having the project totally unusable after the event.

commit 3929f83e (2022-01-11): (UI) Follow up on #351: indexes now handled in by the GraphInterfaceObserver (which will probably be renamed GraphObserver).
Delay actor are now forced to the end of the vertex list.

Might also fix #91 and #122.
closed 2022-02-22 by jheulot