preesm/preesm #96: Problems Import/Export an SDF3 Graph

Arka2009, 2019-01-09:
Hi All,

I am new to PREESM, so while walking through Tutorial-8 "Import/Export an SDF3 Graph". I encounter the following problems. Also, there are no *.graphml generated in Algo/generated folder. The transcript for the workflow execution is as follows

"
13:15:01 NOTICE: Starting workflow execution: /org.ietr.preesm.tutorials.tutorial1/Workflows/SDF3ImportAndListMapper.workflow.
13:15:25 NOTICE: Starting workflow execution: /org.ietr.preesm.tutorials.tutorial1/Workflows/SDF3ImportAndListMapper.workflow.
13:15:25 NOTICE: Workflow Step: Importing SDF3 Xml File
13:15:25 NOTICE: Workflow Step: HSDF Transformation.
13:15:25 NOTICE: Transforming application h263encoder to HSDF
13:15:25 NOTICE: motion_estimation x1
13:15:25 NOTICE: mb_encoding x99
13:15:25 NOTICE: vlc x1
13:15:25 NOTICE: mb_decoding x99
13:15:25 NOTICE: motion_compensation x1
13:15:25 NOTICE: Minimize special actors
13:15:25 NOTICE: HSDF transformation complete
13:15:25 NOTICE: HSDF with 204 vertices and 402 edges.
13:15:25 NOTICE: HSDF transformation: 00:00:00.028s.
13:15:25 NOTICE: Workflow Step: Exporting algorithm graph
13:15:25 NOTICE: Workflow Step: Mapping/Scheduling
13:15:25 NOTICE: Converting from SDF to DAG.
13:15:26 NOTICE: Conversion finished.
13:15:26 NOTICE: mapping a DAG with 210 vertices and 405 edges.
13:15:26 NOTICE: creating route steps.
13:15:26 NOTICE: Initializing routing table.
13:15:26 NOTICE: The Abc parameters are: simulatorType=looselyTimed/approximatelyTimed
13:15:26 NOTICE: Infinite homogeneous simulation
**13:15:26 SEVERE: Relative constraint with no operator.[motion_estimation(1)]
13:15:26 SEVERE: Empty operator set for a vertex: motion_estimation. Consider relaxing constraints in scenario.
13:15:26 SEVERE: Empty operator set for a vertex: motion_estimation. Consider relaxing constraints in scenario.**
"

antoine-morvan, 2019-01-22:
Hi, 

I am currently refactoring the whole application to get rid of the SDF graphs in Preesm and focus the dev effort in the PiSDF (or PiMM) model. Therefore some of the functionalities are getting destroyed. The SDF3 flow is one of those. I'll mark the tuto as deprecated with latest version of Preesm and mention to the last version with which the tutos should work.

antoine-morvan, 2019-01-22:
Fixed with https://github.com/preesm/preesm.github.io/commit/c257351d8aac333abed8a9dbc95a605cc92517c7

commit 25f52e73 (2019-04-16): (Codegen) Fix #96
closed 2019-01-22 by antoine-morvan