preesm/preesm #97: Preesm 3.1.3 does not launch on macOS 10.9

jserot, 2019-01-09:
The app, downloaded from ::

https://github.com/preesm/preesm/releases/download/v3.1.3/preesm-3.1.3-macosx.cocoa.x86_64.zip

crashes at launch time 

[image]

antoine-morvan, 2019-01-21:
Hi, 

I just tried on my Mac running Sierra (10.12.6) with java 1.8.0_181 and could not reproduce the issue. Could you provide more information ?

antoine-morvan, 2019-01-23:
By "Eclipse launcher" I mean the underlying Eclipse framework, and especially the first things that run when you start Eclipse. It is possible to update it manually, but even I do not want to try.

I will release a new version of Preesm by the end of the week, hopefully this will solve the issue.

Until then, you can build Preesm on your machine following these instructions : https://preesm.github.io/docs/buildpreesm/

jserot, 2019-01-23:
Ok, let's wait for the next release and cross fingers ;)

antoine-morvan, 2019-01-24:
I just released 3.2.0 : https://github.com/preesm/preesm/releases/tag/v3.2.0
Let me know if things get better...

jserot, 2019-01-24:
_Pas mieux_ :(
Crash log attached.
[crash.log](https://github.com/preesm/preesm/files/2791064/crash.log)

antoine-morvan, 2019-01-25:
Can you run the default Eclipse packages, as this one https://www.eclipse.org/downloads/download.php?file=/technology/epp/downloads/release/2018-12/R/eclipse-java-2018-12-R-macosx-cocoa-x86_64.dmg ?

antoine-morvan, 2019-01-25:
If we track the Eclipse updates in the release notes, we have:
 * Preesm v3.2.0: Eclipse 2018-12 (4.10)
 * Preesm v2.13.0 : Eclipse Photon (4.8)
 * Preesm v2.4.2 : Eclipse Oxygen (4.7)
So you can try going back in time using those versions. They are all available here : http://preesm.insa-rennes.fr/releases/
Although there has been some updates in the model since then, it should not impact basic examples.

antoine-morvan, 2019-01-25:
I'll close this issue since it's not directly related to our code base.

closed 2019-01-25 by antoine-morvan