preesm/preesm #100: Memory not allocated for small example

ahonorat, 2019-01-17:
In a dummy example with all rates equal to 1, composed of:
- one input actor, sending data to ...
- a broadcast actor, sending data to ...
- one output actor

the memory is not allocated at all in the generated code. This occurs when using memory optimization on broadcast and an architecture of 4 cores. However if more than one output actor are presents, there is no problem.

Furthermore, there is a main thread index error since ```#define _PREESM_MAIN_THREAD_ -1``` is generated in main.c, while the only core used (and generated) is Core1.

antoine-morvan, 2019-01-22:
Please provide (and attach) a complete and if possible minimal example to reproduce.

ahonorat, 2019-02-19:
Here is a MWE: [issue100.zip](https://github.com/preesm/preesm/files/2879900/issue100.zip)

I am pretty sure that the problem is due to the constraints in the scenario: Core1 appears before the other cores although Core0 is the main operator. All actors are allowed on all cores. The scheduler allocates everything on the first core specified in the scenario file (here Core1), and thus the code for Core0 that should contain the memory declarations is not generated.

leos313, 2019-07-01:
Can this bug come from the same reason in [this issue](https://github.com/preesm/preesm/issues/162)?

ahonorat, 2019-08-01:
Seems to be working now, probably since the new parser of scenario.

closed 2019-08-01 by ahonorat