preesm/preesm #106: Code Generation related to the cores seems wrong considering scenario with Papify

ClaudioRubattu, 2019-01-30:
I obtain a C code for each core without any functions related to Papify.

Also, in the main, there is the following code:
#ifdef _PREESM_MONITOR_INIT
    mkdir("papify-output", 0777);
    event_init_multiplex();
#endif
and after the compilation, I can run it, directory has been created, but there is no file inside after the execution (nothing has been monitored).

It seems the issue comes from getCorePapifyConfigGroupActor (present in [preesm/plugins/org.preesm.model.scenario/src/org/preesm/model/scenario/papi/PapifyConfigManager.java](https://github.com/preesm/preesm/blob/master/plugins/org.preesm.model.scenario/src/org/preesm/model/scenario/papi/PapifyConfigManager.java)) that returns null always.

dmadronal, 2019-01-30:
I can't replicate the behavior. 

I am working on the develop branch (updated 10 mins ago) with a standar sobel app and it is working properly for 1 and 2 Cores generating C code (i.e., not SPiDER).

Could you upload (or send me) a project with the problem?

ClaudioRubattu, 2019-01-30:
I have solved using almost the same workflow as in the tutorial.

The issue seems due to the different tasks that I was considering for the DAG generation (org.ietr.preesm.experiment.pimm2srdag.StaticPiMM2SrDAGTask) and for the scheduling (org.ietr.preesm.plugin.mapper.listschedulingfromdag). 

Thank you very much for the availability

ClaudioRubattu, 2019-01-31:
I reopen the issue because, as I said in the previous comment, it seems that the code generation works well with the new workflows but not with the old ones.

dmadronal, 2019-01-31:
Could you send me the workflow that is not working?

ClaudioRubattu, 2019-01-31:
Sorry! Here is the workflow:
[CodegenNew_Papify.zip](https://github.com/preesm/preesm/files/2817055/CodegenNew_Papify.zip)

leos313, 2019-03-21:
I found the same problem: it is not enough to replicate all the step of the tutorial. It is necessary to use the same workflow of the tutorial (with the same tasks in it). Even a little change in the workflow MAY bring to Papify code generation failure.
An example: if you try to add the Papify Engine Task in the following Workflow, you will not see the Papify instruction printed in the generated code. Addictionally, no error is reported. It just does not print.

[image]

dmadronal, 2019-03-21:
I will try to fix it ASAP.

I believe that something related to actor paths is changed for the inputs of the PAPIFY task but I have to check and fix it.

dmadronal, 2019-03-21:
I checked and the incompatibility comes from the new task "org.ietr.preesm.plugin.mapper.listschedulingfromdag".

In the previous task "org.ietr.preesm.plugin.mapper.listscheduling", the path of the actors is not changed even if there are more than one instance of the actor. Example of input paths (null corresponds to explode and implode):

```
sobel_morpho/Read_YUV
sobel_morpho/display
sobel_morpho/Sobel
sobel_morpho/Sobel
sobel_morpho/Sobel
sobel_morpho/Sobel
sobel_morpho/Split
sobel_morpho/Dilation
sobel_morpho/Dilation
sobel_morpho/Dilation
sobel_morpho/Dilation
sobel_morpho/Erosion
sobel_morpho/Erosion
sobel_morpho/Erosion
sobel_morpho/Erosion
null
null
```

Nevertheless, with the new scheduling task "org.ietr.preesm.plugin.mapper.listschedulingfromdag", the paths are changed and, if several instances of the actor are set, an index is added. Example for the same project, changing the workflow:

```
sobel_morpho/Read_YUV_0
sobel_morpho/display_0
sobel_morpho/Split_0
sobel_morpho/Dilation_0
sobel_morpho/Dilation_1
sobel_morpho/Dilation_2
sobel_morpho/Dilation_3
sobel_morpho/Erosion_0
sobel_morpho/Erosion_1
sobel_morpho/Erosion_2
sobel_morpho/Erosion_3
sobel_morpho/Sobel_0
sobel_morpho/Sobel_1
sobel_morpho/Sobel_2
sobel_morpho/Sobel_3
sobel_morpho/explode_Split_0_output
sobel_morpho/implode_display_0_y
```

Is this done on purpose? If so, I will addapt the code to support this change.

dmadronal, 2019-03-21:
I have just pushed a patch to generate working code with PAPIFY (the branch is calledfixPapifyIncompatibility).

It is printing the code properly but there is still one difference when using both schedulings:

For the listscheduling, the vertex.getID() returns the instance name that is set in algorithm UI, while vertex.getName() returns the same instance name BUT adding an index if necessary.

In the listschedulingfromdag, both are returning the instance name adding the index even if there is only 1 instance of the actor.

In this situation I can't merge the results of PAPIFY under the same label (I was using the getID() function to do so) so we will get the metrics splited and probably making the data representation with PAPIFY-VIEWER a bit messy.

What do you think we should do?

antoine-morvan, 2019-03-22:
*SchedulingFromDAG is deprecated and will be removed eventually.

dmadronal, 2019-03-22:
Ok ok.

Then the patch I uploaded does not need to be merged.

Could you send me am up-to-date workflow to check if there is any incompatibility with the PAPIFY task? I mean, a workflow using the tasks that will NOT be removed in the short-term.

antoine-morvan, 2019-03-22:
Latest version of the tutorials have up-to-date workflows : https://preesm.github.io/tutos/parasobel/

dmadronal, 2019-03-22:
I just pushed a new version compatible whit the up-to-date workflow.

The behaviour of returning the same value form vertex.getId() and vertex.getName() happens also in this workflow.

I managed to unify the output for all the workflows with a small workaround that allows me to check if the path of the actor has changed or not.

I would consider this bug closed and I would also encourage you to analyze if the behaviour explained before is expected or not.

commit 0f3b232c (2019-03-22): [RELENG] Update release notes (fix #106);
closed 2019-03-22 by antoine-morvan