preesm/preesm #109: Uncorrect printing in the pgantt file

ClaudioRubattu, 2019-02-07:
Using the plugin "org.ietr.preesm.stats.exporter.StatsExporterTask" to export the pgantt file, the node "perfs" does not contain its attributes. 
Indeed, the symbol ">" is put before them, just after the node name.

antoine-morvan, 2019-02-13:
Please attach a project (or better, a minimal example) so we can try to reproduce.

ClaudioRubattu, 2019-02-13:
The attached project is simply the stereo matching application proposed in [https://github.com/preesm/preesm-apps/tree/master/stereo/org.ietr.preesm.stereo](url) with the adding of the workflow "Codegen_Papify" (that contains the task named "Gantt Exporter" that considers the plugin cited). 

In order to reproduce the error, it is sufficient to run this workflow with one of the present scenarios (for instance "1core" or "4core") and see the generated file in the directory Gantt (that you can find at the root level). In the archive, a Gantt file associated to the run of the "1core" scenario is already present in that folder. 

[stereo.zip](https://github.com/preesm/preesm/files/2859749/stereo.zip)

commit adcd027e (2019-04-09): Fix #109
closed 2019-04-09 by antoine-morvan