preesm/preesm #110: Preesm with Sniper-7.1 on ubuntu18.04 multicore error

eliodilar, 2019-02-08:
Hello,

i am engineer for UBs Lorient Lab-sticc, working on Nooman research project for notifing memories.
i try to supervise Preesm(dataflow) "https://preesm.github.io/" code generated on sniper 7.1, on ubuntu 18.04 with pin 3.7-97619.
i have succefully realise the fft and fork test on gainestown core.cfg(-c gainestown.cfg) with 4 core (-n 4).

After, i have put de the file generated by preesm ( sending data beetwen core on shared memory), i have an error when i want to supervise core -> -n 4.

./run-sniper -c gainestown.cfg $(pwd)/introduction       <- no problem for simulation but snipe only one core.

./run-sniper -n 2 -c gainestown.cfg $(pwd)/introduction <- sim.out sift-recorder impossible...)

I have also a warning on &shared_memory location.

shell return:
SNIPER] Warning: Unable to use physical addresses for shared memory simulation.
[SNIPER] Start
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Sniper using SIFT/trace-driven frontend
[SNIPER] Running full application in DETAILED mode
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Enabling performance models
[SNIPER] Setting instrumentation mode to DETAILED
[RECORD-TRACE] Using the Pin frontend (sift/recorder)
[SIFT_RECORDER] sift_writer.cc:694: bool Sift::Writer::Emulate(Sift::EmuType, Sift::EmuRequest&, Sift::EmuReply&): Assertion `!response->fail()' failed.
[SIFT_RECORDER] sift_reader.cc:180: bool Sift::Reader::Read(Sift::Instruction&): Assertion `!input->fail()' failed.
Pin app terminated abnormally due to signal 6.
[SNIPER] End
[SNIPER] Elapsed time: 5.80 seconds

Can you help me?

Cordially

i have also that problem on sobel generator

[SNIPER] Warning: Unable to use physical addresses for shared memory simulation.
[SNIPER] Start
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Sniper using SIFT/trace-driven frontend
[SNIPER] Running full application in DETAILED mode
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Enabling performance models
[SNIPER] Setting instrumentation mode to DETAILED
[RECORD-TRACE] Using the Pin frontend (sift/recorder)
0          [  834]  [ 0]  [scheduler_pinned_bas: 112]  *ERROR* Invalid core 3 found in sched_setaffinity() mask
[SIFT_RECORDER] si[SI[FST_REIFCOTR_DERRE]C OsRiDERft]_ wrsitiefrt._cwcri:ter.cc:493:49 3u: iunitnt64_t Sift::Writer::Syscall(uint16_t, const char*, uint32_t): Assertion `!response->fail()' failed.
[SNIPER] End
[SNIPER] Elapsed time: 34.85 seconds
Pin app terminated abnormally due to signal 6.
chapoul@lo-lester-007:~/sniper-7.115:14:45$ ^C
chapoul@lo-lester-007:~/sniper-7.115:15:23$ ./run-sniper -n 4 -c gainestown.cfg $(pwd)/sobel
[SNIPER] Warning: Unable to use physical addresses for shared memory simulation.
[SNIPER] Start
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Sniper using SIFT/trace-driven frontend
[SNIPER] Running full application in DETAILED mode
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Enabling performance models
[SNIPER] Setting instrumentation mode to DETAILED
[RECORD-TRACE] Using the Pin frontend (sift/recorder)

[SNIPER] Internal exception: Segmentation fault. Access Address = 0x10

[SNIPER] End
[SNIPER] Elapsed time: 38.73 seconds
------------------------------------------------------------
Segmentation fault. Access Address = 0x10

Backtrace:
    subsecond_time.h:operator<:267
    memory_manager.ccarametricDramDirectoryMSI::MemoryManager::sendMsg:550 (discriminator 2)
    cache_cntlr.ccarametricDramDirectoryMSI::CacheCntlr::insertCacheBlock:1499
    cache_cntlr.ccarametricDramDirectoryMSI::CacheCntlr::handleMsgFromDramDirectory:1772
    memory_manager.ccarametricDramDirectoryMSI::MemoryManager::handleMsgFromNetwork:470
    memory_manager_base.cc:MemoryManagerNetworkCallback:55
    network.cc:Network::netPullFromTransport:120
    sim_thread.cc:SimThread::run:40
    pthread_thread.ccthreadThread::spawnedThreadFunc:21
    ??:??:0
    ??:??:0

kdesnos, 2019-02-08:
Unless I'm mistaken, the issues and error messages related in this issue all originate from Sniper and not from Preesm.

To the best of our knowledge, the code generated with preesm is fully functional for "real" x86 architecture running Linux. Maybe contacting the SNIPER team would be more helpful to solve the above issue.

closed 2019-02-08 by kdesnos

eliodilar, 2019-02-08:
ok, X86_64 also?

Cordially

kdesnos, 2019-02-08:
Yes

eliodilar, 2019-02-08:
ok thanks