preesm/preesm #111: SEVERE: Could not write logs

alemehghasemi, 2019-02-11:
Hello,
I just changed the archi to 4 cores following all the steps but after running workflow I saw this message:
SEVERE: Could not write logs
Is it a serious error?

alemehghasemi, 2019-02-11:
Yes, In fact by this error my new senario has not been applied :(

kdesnos, 2019-02-12:
Hi,
Can you please provide a copy of the project causing this issue.

alemehghasemi, 2019-02-12:
Dear Karol,
hello,
First of all thank you for your reply.
I have solved it because of that I just closed the issue and I could not
clean the post.
it is ok now.
But i have a question:
Do you use some mapping optimization for power consumption or something
like this?Because when I assign actors over cores sometimes I see that in
guntt chart some cores are omitted.

Thank you so much for your consideration.
Alemeh,

kdesnos, 2019-02-12:
An actor can be mapped to any core, following the mapping constraints specified in the scenario.

When changing the number of cores in the architecture description, just don't forget to update the scenario accordingly to enable the execution of actors on these new cores, as explained in the tutorials.

Another reason why a core may receive no actor execution is if the scheduling algorithm finds a faster solution without assigning any work on these cores. This may happen if an application does not have enough parallelism, or if a core of the architecture is much slower than the others, making it highly uninteresting to put some work on this core.

closed 2019-02-12 by kdesnos

alemehghasemi, 2019-02-12:
Thank you.
I am just using introduction application of project tutorial ! of Preesm
over a homogeneous 8cores-1shared memory architecture.
 I assigned every actor to one core (one by one) but as can be seen in the
output result of the sniper, attached, the work load is not balanced at
all. ( "Idle time (%) ")
As core0 is 48% idle and core7 80% and others are almost idle.