preesm/preesm #112: Order of communication primitives (Send/Receive) is not preserved between components

leos313, 2019-02-12:
When having a S-LAM like the one in the following picture: 

[image]

The output of PREESM is:

    ...
    09:40:56 NOTICE: Adding OBN actors to CPN and IBN actors in CPN dominant list
    09:40:56 NOTICE: Mapping
    09:40:59 NOTICE: Mapping finished
    09:41:09 NOTICE: removing 24 syncs
    09:41:09 SEVERE: Order of communication primitives (Send/Receive) is not preserved between components org.preesm.model.slam.impl.ComponentInstanceImpl@1269b861 (instanceName: Core1) and org.preesm.model.slam.impl.ComponentInstanceImpl@48582d00 (instanceName: Core6). Contact Preesm developers for more information.

Idea on what is going on?

antoine-morvan, 2019-02-13:
Please attach the project so we can try to reproduce.

leos313, 2019-02-13:
This [matrix_multiplication_test.tar.gz](https://github.com/preesm/preesm/files/2859730/matrix_multiplication_test.tar.gz) is a clean version of the project that produces the error. 

**Further information**:
 * Also the last release of PREESM-3.4.0 gives the same results.
 * The Parameter `dim_div_factor` of the *PiSDF* is critical: when set up to 2 there are no problems; when set up to 4 (like in the attached example), PREESM has the log that I reported:

```
...
09:40:56 NOTICE: Adding OBN actors to CPN and IBN actors in CPN dominant list
09:40:56 NOTICE: Mapping
09:40:59 NOTICE: Mapping finished
09:41:09 NOTICE: removing 24 syncs
09:41:09 SEVERE: Order of communication primitives (Send/Receive) is not preserved between components org.preesm.model.slam.impl.ComponentInstanceImpl@1269b861 (instanceName: Core1) and org.preesm.model.slam.impl.ComponentInstanceImpl@48582d00 (instanceName: Core6). Contact Preesm developers for more information.
```

kdesnos, 2019-02-19:
A simpler project to reproduce the issue (after a long investigation) : 
[failingCommuicationOrder.zip](https://github.com/preesm/preesm/files/2880801/failingCommuicationOrder.zip)

The problem seems to occur in very (very) specific conditions:
- Two actors (explode_A_0... and explode_B_... actors in the example) generate enough data to trigger more than 11 (i.e. 12 and more) firings of an actor C.
- The actors generating the data (explode_A_0... and explode_B_...) are executed on two different cores. If they're executed on the same core, the issue does not arise.
- The cores on which the actors generating data (explode_A_0... and explode_B_...) are executed are BOTH linked to the core on which the actor consuming the data is executed with a multi-step route (i.e. communication goes through 2 comNode & one PE before reaching the destination core).

Issue may (or not) be coming from reordering code for send receive vertices in org.preesm.algorithm.mapper.abc.transaction.AddSendReceiveTransaction