preesm/preesm #116: [RELENG] Compile errors remain after following all steps

antoine-morvan, 2019-02-18:
With latest versions, some compilation errors remain in the workspace despite following all the steps in troubleshooting. The exact cause is not identified yet.

Temporary solution consists in:
- follow all steps normally until https://preesm.github.io/docs/buildpreesm/#compilation-of-preesm
- When build ends, errors remains. There, locate the project **org.preesm.commons** and Right click / Close project;
- Wait for the build process to finish, then re-open the project.
- The build will start again, but this time removing all errors.

antoine-morvan, 2019-03-29:
Eclipse bug report: https://bugs.eclipse.org/bugs/show_bug.cgi?id=545931

hmiomand, 2025-01-25:
Fixed.

closed 2025-01-25 by hmiomand