preesm/preesm #117: 11:14:28 SEVERE: Missing task parameters: the task Scheduling requires parameters (Optimize synchronization).

alemehghasemi, 2019-02-19:
Hello,

I am trying to run workflow of Odroid preesm application. 
I have set all the parameters in TestCom2_Odroid.senario as can be seen in pictures attached .
But still I have error while running Codegen.workflow and RunAnalysis.workflow.

Thank you in advance.
Alemeh,

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

antoine-morvan, 2019-02-19:
Hello, 

Please read latest new https://preesm.github.io/release-3.3.0/

closed 2019-02-19 by antoine-morvan

alemehghasemi, 2019-02-19:
Thank you and then I received these errors. 

it's looking for an excel file:
SEVERE: The custom quanta input file /org.ietr.preesm.experiment.odroid.moa1/stats/mat/custom_quanta_in/quanta_in_TestCom2_Odroid.xls does not exist.

14:07:45 SEVERE: Unexpected Exception: java.lang.NullPointerException:null
 Contact Preesm developers if you cannot solve the problem.

alemehghasemi, 2019-02-19:
I think it is a timing setting issue I have changed the timing excel file but nothing happened.
how can we set quanta value? just with excel file?

kdesnos, 2019-02-19:
Hi Alemeh,
Are you working on the `org.ietr.preesm.experiment.odroid.moa1` project for a particular reason. (If this was discussed with Maxime, I'm not aware of it).
As indicated in the name of the project, this project and workflow are highly experimental, and aim at testing ideas around Models of Architectures in Preesm.

So unless you are working with the MoAs, maybe it'd be better to switch to less experimental projects.
If however you ARE working with MoA, we'll redirect your questions to Maxime :)

alemehghasemi, 2019-02-19:
Hello Karol,
No, in fact he main reason behind choosing this application is that I need a more complicated application with further number of actors. just for having more traffic and work load to distribute over the cores.
I would appreciate if you let me know whether there is any application among preesm apps (reinforcement learning, stabilization, adaptative runtime, stereo) with this features that I can run successfully.
Thank you in advance.

alemehghasemi, 2019-02-19:
Hi Karol,

I successfully run Stabilization app.So I will use this one instead.
Thank you for your help.
Alemeh,

alemehghasemi, 2019-02-19:
Sorry karol for asking again, now for running executable file of stabilization I need jaguarandcroco360x202-375f.yuv file. A
[image]
s can be seen in error there is no such file there.

alemehghasemi, 2019-02-19:
I just rename input files of sobel an give them to this application it run but the output video is wrong.
I would appreciate if you let me know from where I can download arial.ttf and JaguarAndCroco360x202-375f.yuv
Thank you so much.

kdesnos, 2019-02-20:
I think all needed info is available in the Readme file in the following folder:
https://github.com/preesm/preesm-apps/tree/master/org.ietr.preesm.stabilization/Code/dat

alemehghasemi, 2019-02-21:
Thank you so much Karol.

alemehghasemi, 2019-02-25:
Hello Karol,

Apparently Preesm put the display video inside a loop.
How can I change the generated code to cut the loop and generated code just
display original time for instance 15 seconds for Jaguar mp4?
Thank you in advance.

Best Regards,
Alemeh,

antoine-morvan, 2019-02-25:
Please continue with mail or open a new issue.