preesm/preesm #119: Cutting the loop in generated code

alemehghasemi, 2019-02-25:
Hello,

Apparently Preesm put the application inside an infinite loop.
How can I change the generated code to cut the loop so that it just displays the video in its original time for instance for Jaguar mp4, 15 seconds?
Or maybe I should change timing ?
Thank you in advance.

Alemeh,

antoine-morvan, 2019-02-25:
Hi, 

In the user header **preesm.h**, you can define the preprocessor constant ```LOOP_SIZE```.

For the video applications, each iteration of the loop will display 1 frame. So for 15s at 30 fps, you will have to set this loop size to 15*30 = 750
```
#define LOOP_SIZE 750
```

alemehghasemi, 2019-02-25:
Thank you Antoin for your help.

closed 2019-04-15 by antoine-morvan

alemehghasemi, 2019-04-16:
> SInce 3.6.2, macro has been prefixed with *PREESM_* to avoid possible
> conflicts with application macros.
>
> Also documentation on macros in now available in the task reference:
> https://preesm.github.io/docs/workflowtasksref/#code-generation-1
>
> See https://preesm.github.io/release-3.6.2/

Thank you so much Antoine for your help.

commit 42894b19 (2019-04-15): Merge branch 'fix119' into develop

Fix #119