preesm/preesm #120: Latency increases enabling more processing elements (PEs)

ClaudioRubattu, 2019-03-11:
I have verified a latency increasement considering the attached stereo matching application. 
You can take a look at the Gantt charts present in the directory "Gantt". In particular, in the same folder, there is "stereo_parallel_factors.csv" that summarizes the execution time for each configuration of the choosen architecture. As you can see, considering 3 PEs leads to a bigger latency (317,278,398 cycles) than the case with 2 ones (295,833,934 cycles).  Could it be possible?

[stereo.zip](https://github.com/preesm/preesm/files/2951733/stereo.zip)

antoine-morvan, 2019-03-13:
After testing a bit, it seems the memory copy speed is set to 1 in your scenarios. That means it takes 1 cycle to copy 1 byte.
The list scheduler is probably not aware of the cost of such communications. Increasing the memory copy speed in the scenario will solve this.

When executing the application, what were the measured execution times?

antoine-morvan, 2019-03-13:
@mpelcat

antoine-morvan, 2019-03-14:
The communication overhead abstraction seems too simple and not properly taken in consideration in the scheduler. This results, as explained in the original post, in higher latency when adding cores. Indeed when the communication cost is too high, the scheduler should put everything on 1 core.

Note: check if this is also related with the "Average Transfer Size" from the Simulation tab in the scenario.