preesm/preesm #122: adding actors to tutorial1, creates a run error.  Removing said actors does not make it go away

LowellHuawei, 2019-03-28:
In 3.5.0, in the algo diagram, I added a disconnected pair of actors connected by fifo.  I set the rates and the periods to 1 for both actors and fifos.  I didn't edit anything else -- for example, I did not edit the scenario.  I did a run workflow and selected the TestCom PC scenario.  I got this error

Unexpected Exception: java.lang.UnsupportedOperationException:Flat single-rate can not have PiSDFRefinement for a vertex.
 Contact Preesm developers if you cannot solve the problem.

So I went back and deleted the actors and the fifos that I had just added -- then did Ctrl-Shift-S to save all.

However, when I tried to repeat the run steps, mentioned above, without my newly added changes, I still get the same error.

LowellHuawei, 2019-03-28:
I neglected to mention that the act of saving, with Ctrl-Shift-S caused this error

platform:/resource/org.ietr.preesm.tutorials.tutorial1/Algo/TestComp.pi
java.lang.NullPointerException

LowellHuawei, 2019-03-28:
Ok, so the java error in the save process was causing TestCom.pi not to be correctly written -- which meant that during the run workflow stage, there were  non-existent items in the workflow.  I was able to hand edit the TestCom.pi file and run the workflow without errors.

So basically, if you edit the TestCom.pi file and ignore the Algo graphical editor,  it works ok.

Is there a adocument that explains all the tags of interest in the .pi files?  I'll look but don't recall seeing one before.

antoine-morvan, 2019-04-02:
The first error you encountered was because the 2 actors you added did not have any refinement (C or hierarchical graph). Thus when the task that transforms the PISDF into the scheduling intermediate representation fails. The exception thrown there is wrong and should print a more accurate message.

The issue with saving the graph is probably linked to the issue #91 ... for which we have a hard time reproducing and fixing.

closed 2023-01-25 by hmiomand