preesm/preesm #125: cli cannot access org.eclipse.templates.freemarker.javall11

LowellHuawei, 2019-04-01:
I have turned off my firewall but am running into trouble running commandLinePreesm.sh.  It cannot access org.eclipse.templates.freemarker.java11 . Is there a trouble shooting guide?

I checked on the internet for a solution, see the bottom of this issue for details and a question.

Here's what happened:

Note that I ran cygwin bash from my PATH -- I did not copy it underneath the PREESM directory

I'm trying to use the command line cli on an existing project that builds fine when I use eclipse.  When I am in cygwin bash, and run this command

./commandLinePreesm.sh  'C:/Users/l84115432/preesm/3.5.0' C:/Users/l84115432/preesm/preesm/nextTry.zip_expanded/lowell-copy-tutorial1 Codegen.workflow TestComPC.scenario

I get this error:
***START*** Mon, 01 Apr 2019 10:06:32 -0500

Init workspace and import project

Saving workspace.
!SESSION 2019-04-01 10:06:33.202 -----------------------------------------------
eclipse.buildId=unknown
java.version=1.8.0_201
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -application org.eclipse.cdt.managedbuilder.core.headlessbuild -import C:/Users/l84115432/preesm/preesm/nextTry.zip_expanded/lowell-copy-tutorial1
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -consolelog -data C:/Users/L84115~1/AppData/Local/Temp/tmp.B4jwCnsPEQ_preesm-workspace -application org.eclipse.cdt.managedbuilder.
build -import C:/Users/l84115432/preesm/preesm/nextTry.zip_expanded/lowell-copy-tutorial1

!ENTRY org.eclipse.tools.templates.freemarker.java11 2 0 2019-04-01 10:06:35.245
!MESSAGE Could not resolve module: org.eclipse.tools.templates.freemarker.java11 [258]
  Unresolved requirement: Fragment-Host: org.eclipse.tools.templates.freemarker; bundle-version="1.0.0"

!ENTRY org.eclipse.osgi 4 0 2019-04-01 10:06:35.256
!MESSAGE Application error
!STACK 1
org.eclipse.core.runtime.CoreException: No file system is defined for scheme: C
        at org.eclipse.core.internal.filesystem.Policy.error(Policy.java:48)
        at org.eclipse.core.internal.filesystem.Policy.error(Policy.java:43)
        at org.eclipse.core.internal.filesystem.InternalFileSystemCore.getFileSystem(InternalFileSystemCore.java:68)
        at org.eclipse.core.internal.filesystem.InternalFileSystemCore.getStore(InternalFileSystemCore.java:110)
        at org.eclipse.core.filesystem.EFS.getStore(EFS.java:481)
        at org.eclipse.cdt.managedbuilder.internal.core.HeadlessBuilder.importProject(HeadlessBuilder.java:338)
        at org.eclipse.cdt.managedbuilder.internal.core.HeadlessBuilder.start(HeadlessBuilder.java:487)
        at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
        at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
        at org.eclipse.equinox.launcher.Main.run(Main.java:1501)

This web page
  https://bugs.eclipse.org/bugs/show_bug.cgi?id=542740

Suggests that I upgrade to java 11 -- should I ?

Here's what it says:

Alexander Kurtakov                                        2018-12-13 05:29:15 EST                That is to be expected as it has been added to overcome the issue of Java 11 removing JAXB. See https://git.eclipse.org/c/cdt/org.eclipse.tools.templates.git/commit/?id=dc437804a238f57a12675f882ba2fbac7c79ef1a for details.
--

Thanks for any advice you can give

antoine-morvan, 2019-04-02:
Hi, 

Please use backslashes in your command when using Windows:

```bash
./commandLinePreesm.sh "C:\Users\l84115432\preesm\3.5.0" "C:\Users\l84115432\preesm\preesm\nextTry.zip_expanded\lowell-copy-tutorial1" Codegen.workflow TestComPC.scenario
```

closed 2019-04-02 by antoine-morvan

LowellHuawei, 2019-04-02:
I’ll try it!  Thanks!

LowellHuawei, 2019-04-02:
Thanks, the suggestion from Antoine worked!  My problem was related to the backslashes.

I made my own copy of commandLinePreesm.sh and changed the variable assignments at the top like this:

#!/bin/bash -eu

## the path to the Preesm binary directory
PREESMDIR=$( echo ${1} | tr '/' '\\' )

## the folder containing the application to run (where the .project is)
APPDIR=$( echo ${2} | tr '/' '\\' )

## workflow and scenario to run (must be respectively under Workflows and Scenarios folder within the project)
WORKFLOW=$( echo ${3} | tr '/' '\\' )
SCENARIO=$( echo ${4} | tr '/' '\\' )

This eliminates the need for manual conversions of path and still seems to work.  Switching in and out of bash and the windows command line happens a lot for me, it’s hard to keep everything straight by hand.

Regards,
   Lowell

antoine-morvan, 2019-04-02:
Please note that using such way (tr to '\\') might break script compatibility with Unix systems.

LowellHuawei, 2019-04-02:
Oh yeah, thanks.  I’ll have to modify the script to check for “windir”, “programfiles” or other windows environment variable before making the substitution.

Thanks!

LowellHuawei, 2019-04-02:
Sorry, you have to check for

   If [ “$OS” = “Windows_NT” ]

Because most of the windows variables don’t show up as visible in the eclipse run of the shell script.

Regards,
   Lowell