preesm/preesm #126: an error while parsing 64coreX64.slam and segmentation fault for generated application

alemehghasemi, 2019-04-02:
Dear all,
Hello,

I would appreciate if you can help me for the following problem:
I was faced with a problem during generating new code for new archi.
new created archi:
64 x64cores, in which all them are connected to one shared memory. 
I successfully generated the gantt, but when I tried to run executable file of application I received "segmentation file".
Backing to change shared memory speed in archi file. I could not open .slam file any more.
You can find attached these errors:

Regards,
Alemeh,

[image]
[image]

kdesnos, 2019-04-02:
Dear Alemeh,
Can you please provide a minimal project to reproduce the issue.
Best,
Karol

alemehghasemi, 2019-04-11:
Dear Karol,
I have generated code over an archi with 32 cores and one shared memory.
I generated the gantt successfully multiple times and I did not have that parsing error with .slam file but still segmentation fault while running the application! (following error):
-----------------------------------------------------------
SDL_Init_Start
SDL_Init_end
TTF_Init
register
Segmentation fault (core dumped)
-----------------------------------------------------------------------
I tried with tutorial1 application as a minimal one but I could not have all 32 cores in gantt and I tried many types of mapping but I think Preesm try to eliminate idle redundant cores in case of simple application. 
I should add that I successfully generated this stabilization with 4, 8, 12, 16 threads (core) in Preesm and did not have any problem while running the generated application.
Thank you for you time and help in advance.
Best,
Alemeh,

kdesnos, 2019-04-11:
Dear Alemeh,
Can you please provide a copy of the problematic project to reproduce the issue.
Best,
Karol

alemehghasemi, 2019-04-12:
Thank you for your help.
you can find attached the zip folder of problematic project.
Best Regards,
Alemeh,
[org.ietr.preesm.stabilization-segmentation-fault.zip](https://github.com/preesm/preesm/files/3072157/org.ietr.preesm.stabilization-segmentation-fault.zip)

antoine-morvan, 2019-04-12:
Could reproduce, looks like an non proper initialization of synchronization mechanism.

alemehghasemi, 2019-04-12:
You mean reproducing the archi and the scenario file?

antoine-morvan, 2019-04-12:
In the generated code, in the file **communicatin.h**, there is a constant that defines the maximum number of cores. It is used to initialize the semaphore array. The code generation of Preesm hardcode it to 16:
```C
#define MAX_NB_CORES 16
```
It has to be manually overridden after every code generation to the actual number of cores if there are more than 16.

More user friendly solution would be to have this value generated from the SLAM input.

alemehghasemi, 2019-04-12:
Yes, you are right. I run it successfully now!
Thank you Antoine for your help.

commit 4247af5e (2019-04-12): (Codegen) Add constants; fix #126
closed 2019-04-15 by antoine-morvan