preesm/preesm #129: tycho pom jar no longer available

LowellHuawei, 2019-04-05:
**Where can I find the typco-maven-plugin/1.3.0?**

I am using eclipse photon for java and the PDE, and have imported the needed predecessors for preesm, and have attached my eclipse to the 3.5.0 developer version of the preesm repository.

The build seems to work for everything except the POM files.  every single one of them fails.

In my  c:/users/userid.m2 directory, down the repository/org/eclipse/tycho/tycho-maven-plugin/1.3.0/tycho-maven-plugin-1.3.0.pom.lastUpdated file, I find that this error message:

http\://repository.jboss.org/nexus/content/repositories/releases/.error=Could not transfer artifact org.eclipse.tycho\:tycho-maven-plugin\:pom\:1.3.0 from/to sonatype-public http\://repository.jboss.org/nexus/content/repositories/releases/)\: Permission denied\: connect

**Yes, I have done web searches looking for the tycho jar**

One place it told me to look was http://repository.jboss.org/nexus/content/repositories/releases/org/  but there is no **tycho**  directory found there.  Nor is there an "**org**" or an "**eclipse**" or a "**maven**".

I tried installing m2e but it told me that it is already installed.

I also tried going to http://m2eclipse.sonatype.org/ but the web site no longer exists.

Any ideas?

thanks,
   Lowell

antoine-morvan, 2019-04-08:
Hi,

typho-maven-plugin should be automatically donwloaded by Maven or Eclipse M2E plugins. Maven plugins can be searched on various dedicated search engines, as this one : https://mvnrepository.com/artifact/org.eclipse.tycho/tycho-maven-plugin

Note that although not advised, it is possible to manually install a jar in the local Maven repository. See here : https://maven.apache.org/guides/mini/guide-3rd-party-jars-local.html

The errors you have seems to indicate a corrupted local Maven repository. The `Permission denied: connect` seems to indicate that either your firewall blocked the access, or the website rejected you (down times happen pretty often when using Eclipse Update Sites or Maven repositories), or maybe your filesystem rejected the write.

First, i would suggest to:
1. stop eclipse
2. delete `.m2/repository/org/eclipse/tycho/*` (the assumed corrupted part of the repo)
3. start eclipse
4. select all the Preesm related projects in the workspace, and press Alt+F5 then press OK.

This should trigger a new fetch of tycho plugins.

You can also go through the [Trouble Shooting](https://preesm.github.io/docs/buildpreesm/#troubleshooting) of our build doc.

I will try to reproduce in the meantime.

Also, why did you chose Eclipse Photon over latest releases ?

Best.

LowellHuawei, 2019-04-08:
Thanks for the detailed response.

Is there a preferable version of eclipse?  I just took the latest one.

I will try to fix photon then roll back if I am still having trouble.

Thanks again!

Lowell

antoine-morvan, 2019-04-08:
Latest release is not **Photon** but **2019-03**.
See
  * https://wiki.eclipse.org/Simultaneous_Release
  * https://en.wikipedia.org/wiki/Eclipse_(software)#Releases

LowellHuawei, 2019-04-08:
Thanks!

  *   https://wiki.eclipse.org/Simultaneous_Release

LowellHuawei, 2019-04-08:
Dear Mr. Morvan,

Thanks you for your help.

I went to the page that you suggested, https://mvnrepository.com/artifact/org.eclipse.tycho/tycho-maven-plugin, and saw that it suggested that I use the CENTRAL repository.  Which is a URL.

I used that URL in the eclipse / help / install new software page.  But it said that there is no software found there.

Should the maven repository URL be used in some other context?

Thanks,
   Lowell

antoine-morvan, 2019-04-08:
Maven Central is a Maven Repository, that hosts Maven Packages (Jar files with associated [POM file](https://maven.apache.org/guides/introduction/introduction-to-the-pom.html)) and Maven Plugins (which are also Maven Packages). This kind of repository is used by Maven to build and fetch dependencies.

Eclipse uses another kind of repository, called "Update Site" or "P2 Repository", that hosts OSGi. Both the layout of the repository and the content of the Jar files differ.

It becomes tricky when you build Maven projects with Eclipse, or Eclipse projects with Maven: indeed Eclipse has to fetch dependencies described in the [POM file](https://maven.apache.org/guides/introduction/introduction-to-the-pom.html) of the projects, and mimic the job of specific plugins. In the same way, Maven needs to know how to build some files that are automatically handled by Eclipse (Xtend files for instance, when proper Eclipse plugins are present).

This "interfacing" is done by 2 things:
  * Interfacing from Maven to Eclipse : this is done by the set of Eclipse plugins "Maven 2 Eclipse" or M2E. The base plugin of M2E read the project configuration, and calls for "connectors" to handle the specific plugins. Part of Preesm configuration tells M2E to ignore some plugins, see this example (`<action><ignore/>...`) : https://github.com/preesm/preesm/blob/01986950a11097824be4240605427f6a3e022775/pom.xml#L774-L835
Discovery of connectors is done automatically by M2E from this catalog (https://repo1.maven.org/maven2/.m2e/discovery-catalog/connectors.xml)
  * Interface from Eclipse to Maven: this has to be done manually, by configuring the POM files properly, see this example : https://github.com/preesm/preesm/blob/01986950a11097824be4240605427f6a3e022775/pom.xml#L181-L198

Back to your problem with the tycho-maven-plugin. This missing Jar file is a Maven plugin. You cannot install it using the Eclipse Update Site mechanism. As stated above, you can either install it manually, install it via Maven command line, or install it via M2E. The later one can be achieved by forcing the M2E plugin to update the Maven configuration of Projects (select projects, alt+f5, ok). This will trigger the fetching of missing Maven plugins/dependencies. However, if your Maven Local repository (user/.m2/repository) is corrupted, this can fail. Thus I suggest to delete the corrupt part and let M2E reinitialized it properly.

LowellHuawei, 2019-04-08:
Thanks for the clarification – and patience.

Lowell

LowellHuawei, 2019-04-08:
Dear Mr. Morvan,

I think that your advice fixed the problem.  I deleted the org/eclipse/tycho directory, then selected all preesm projects in eclipse and pressed Alt-F5 then OK.  It took 3 iterations of doing this, but now it looks like all my errors went away.  I then did a clean build and got this screen which I think means a successful build because the error log is empty and there are no red X beside the project names.

[cid:image001.jpg@01D4EE02.1362E2E0]

Thanks again for your assistance!
Lowell

antoine-morvan, 2019-04-09:
Actually it takes a while to fetch maven dependencies, and the progress bar is really small in the bottom right of Eclipse window:
[image]
[image]
I'll add that in the build documentation.

antoine-morvan, 2019-04-15:
The generated Eclipse product for Preesm is not "C++" based. It is simply a releasing choice to integrate CDT: indeed the Preesm projects revolve around C/C++ files, and having the C editor in the IDE is, in my opinion, a must.

Preesm can run without CDT (remove required features from feature.xml and prodcut). It would also require to build a version of preesm/graphiti without CDT.

antoine-morvan, 2019-04-15:
Doc additions available here: https://preesm.github.io/docs/buildpreesm/#import-preesm-projects

closed 2019-04-15 by antoine-morvan