preesm/preesm #130: how to export a build for my coworkers to use?

LowellHuawei, 2019-04-09:
Some of the problems our team has experienced with the released Preesm builds seem to have gone away in the developer branch version that Antoine helped me build.

How can I export my built Preesm to for my co-workers to use?

antoine-morvan, 2019-04-09:
There is some doc here : https://github.com/preesm/preesm/blob/develop/releng/README.md

It is out dated, but the maven process to build the zip archives should work.

LowellHuawei, 2019-04-12:
this seems to have worked.  Thanks for the assistance.

Perhaps the instructions should say how to find the actual sub-directory where the built executable environment can be found?  I used find . -name eclipse.exe after the build to find the x86_64 directory where the code was deposited.    Also, the build fails do to lack of the server to host the output on -- the sourceforge server.  This is not important to me, but it took me a while to realize that the failure didn't matter.  Perhaps a note about that in the build instructions would help.

But I am closing the issue since it all worked and I have been able to distribute my built executables.  Thanks again for the timely assistance!

antoine-morvan, 2019-04-15:
Well, the doc is out dated, and needs to be rewritten, so I'll let this issue open until then.

closed 2019-04-15 by antoine-morvan