preesm/preesm #131: ERROR: Task read cannot open ppm_file './dat/im0.ppm'

alemehghasemi, 2019-04-15:
Dear all,
Hello again,
I would appreciate if you give me some help to solve following error:
I have generated 4-thread stereo over typical 4-core archi.
I also tried different following mappings:
1- core0: all actors + read actors 
other cores: all actors
2- same as previous mapping but I did not map delay and broadcast actors over cores anyway in scenario-simulation tab, I selected all cores for running br and delay actors.
I have downloaded into dat folder all input files needed but I do not know why I received this following error while running generated code.

./stereo 
readPPMInit()
ERROR: Task read cannot open ppm_file './dat/im0.ppm'
readPPMInit()
ERROR: Task read cannot open ppm_file './dat/im1.ppm'
SDL: could not create texture - exiting
You can find the project attached if it is needed.

Sincerely yours,
Alemeh,

[org.ietr.preesm.stereo-reading-ppm-problem.zip](https://github.com/preesm/preesm/files/3079980/org.ietr.preesm.stereo-reading-ppm-problem.zip)

antoine-morvan, 2019-04-15:
Hi, 

First, the error **cannot open ppm_file './dat/imX.ppm'** obviously tells that you are running the binary from the wrong folder.

Second, in the scenario, you have to make sure the 2 Read and 2 Display actors execute on the same core. Practically, that means disallowing execution of actors **Display_rgb0** and **Display_rgb1** on all cores except core 0. This is an documented behavior of SDL2 (https://wiki.libsdl.org/FAQDevelopment#Can_I_call_SDL_video_functions_from_multiple_threads.3F).

Best.

closed 2019-04-15 by antoine-morvan

alemehghasemi, 2019-04-15:
Thank you Antoine. I always used to run it fro Release folder.
And also thank you for SDL document.
Best,
Alemeh,