preesm/preesm #132: Update timings in Preesm

antoine-morvan, 2019-04-15:
Timings in Preesm are no good:
*  The scenario editor does not allow giving different timings to different instances of an actor. This is an issue since different instances with different parameter values might have different execution time (see https://github.com/preesm/preesm/issues/77);
*  The scenario editor does not allow giving parameterized timings. Although instrumentation might never generated parameterized timings, the user might now the complexity of his own code (see https://github.com/preesm/preesm/issues/73);
*  The scenario editor cannot import instrumentation results as is: the exported CSV must be processed manually in order to import timings properly. This must be fixed (see https://github.com/preesm/preesm/issues/52);
*  The new instrumentation library (dump.c/h) has to be tested on C6678 (see https://github.com/preesm/preesm/issues/95);
*  Timings seem to not be properly propagated to all steps (see https://github.com/preesm/preesm/issues/121);

antoine-morvan, 2019-06-17:
2 first bullets are fixed since https://github.com/preesm/preesm/commit/42b11efd4781881b754a9f05b25a77e9eddf464d.

leos313, 2019-06-25:
Hi Guys,

What is new on this bug? Are the 3 last points still open?

Thanks

antoine-morvan, 2019-06-25:
Points 1 and 2 are fixed in latest release. Alexandre H. is working on points 3 and 4. It's up to the scheduler to use the timing information. The Gantt exporter should be using them as of latest release.

ahonorat, 2019-07-25:
All related issues have been closed, and new timings seem to work (specification, instrumentation, and parsing) since bd8fe33 , thus the issue can be closed.

closed 2019-07-25 by ahonorat