preesm/preesm #135: Defunct process after exiting sobel app on CentOS

dmadronal, 2019-04-16:
I have just configured the environment to work with CentOS and with the basic example of sobel-morpho I got defunct processes and both the application and the terminal are stalling until the process is killed.

This issue appeared when using only 1 core (but I can imagine that it is a general problem). When I either close the display window or do ctrl+c to kill the app, the variable stopThreads becomes 1 and, hence, the pthread_exit(NULL) function existing on Core_0.c is executed.

As the main thread of the application is executing the Core_0 task, there is no thread to finish the application, so it stalls.

Potential fixes: change while(1) by while(!stopThreads) // change pthread_exit(NULL) by break;

antoine-morvan, 2019-04-16:
Could not reproduce on Ubuntu, however described behavior and proposed change make sense.

Using `!stopThread` as loop condition is another alternative.

This issue also denotes the lack of documentation for the stopThread variable. This variable could also be prefixed to avoid conflicts with application variables.

antoine-morvan, 2019-04-16:
Also, using return NULL; instead of pthread_exit(NULL); when running main core code could fix the issue temporarily. Better synchronization is required here.

dmadronal, 2019-04-16:
Yes, all the proposed solutions fix the issue.

commit 85881fb8 (2019-05-22): (Codegen) Rename stopThread; change loop condition

fix #135
commit 09fa766a (2019-05-22): (Codegen) Rename stopThread; change loop condition

fix #135
closed 2019-05-22 by antoine-morvan