preesm/preesm #142: (Codegen) Compatibility with the `DistributedOnly` memory allocation.

kdesnos, 2019-05-03:
Code generation for Distributed Only memory architecture is currently not supported in Preesm.

Indeed, using the code generation in a Workflow where the `Distribution` parameter of the Memory allocation (and Memory Script) task(s) was set to `DistributedOnly`, the following Exception occurs:
>  DistributedOnly distribution policy during memory allocation not yet supported in code generation. DAGEdge <dagEdgeName> is already associated to a Buffer and cannot be associated to a second one."

kdesnos, 2019-05-03:
Here is a potential todo list to implement this feature without modifying the `CodegenModelGenerator` too much. Simply explained, the idea is to create a new TwinBuffer class corresponding to a Buffer that should be allocated and printed twice: once on each side of the communication corresponding to this Buffer.

TODO:
* Create a `TwinBuffer` class in `org.preesm.codegen/model/Codegen.xcore` that
   * Inherits from Buffer (or maybe SubBuffer, but I don't think so)
   * Possesses two attributes of type Buffer (each corresponding to the allocation of the same Buffer, in two different PE)
       * Although these attributes are sometimes referred as "first and second" child (to denotes the order in which the two childs are associated to the TwinBuffer instance), there is no order for thes attributes. (i.e. a Set could be used to store them)
   * The PE on which each (child) Buffer is allocated can probably be retrieved from its `creator` attribute (inherited by the Buffer class from the Variable class)
   * The creator attribute of the `TwinBuffer` class itself may be problematic, as it is not unique. Maybe leaving this attribute undedined when instantiating the class will be sufficient.. since th TwinBuffer itself will never be printed, only its child buffers will.

* In `CodegenModelGenerator.generateBuffer()`:
   * Remove exception from lines `880` to `894` (leave a warning in its place associated to an equivalent issue on Github)
   * In place of this exception, create a call to a generateTwinBuffer() function. Give it the `SubBuffer dagEdgeBuffer` instantiated in line `851` as a param, and the dagEdge `originalDagEdge`
   * Move the call to `generateSubBuffers()` from line 868 to the inside the following if statement (line 877 - 879) 
       ```if (!this.dagEdgeBuffers.containsKey(originalDagEdge)) {
            this.dagEdgeBuffers.put(originalDagEdge, dagEdgeBuffer);
          } else {
	   ```

* In a new `CodegenModelGenerator.generateTwinBuffer(subBuffer, dagEdge)` function 
   * If this is the first call to this function for this dagEdge (i.e. the "other" Meg was not yet processed in generateBuffer() and the dagEdgeBuffers has no entry for the currend dagEdge)
       * Then 
	       * Instantiate the TwinBuffer 
		   * store the TwinBuffer in the dagEdgeBuffers map. 
		   * Store the SubBuffer received as a parameter as a (first) child the TwinBuffer.
		   * call generateSubBuffers.
	   * Else 
	       * retrieve the TwinBuffer from the dagEdgeBuffers 
		   * Store the SubBuffer received as a parameter as a (second) child the TwinBuffer.
		   * call generateTwinSubBuffers with the TwinBuffer and the processed dagEdge as a param.
		   
* In a new `generateTwinSubBuffers(TwinBuffer, DagEdge)` function
        * Like subbuffers of DagEdge in generateBuffer, all subBuffers corresponding to srSdf Edges must be associated to a TwinBuffer. 
		* Ideally, it would be better not to duplicate the code from `generateSubBuffers()`. Hence, my idea would be in the generateTwinSubBuffers function to :
		    * Retrieve and store all SubBuffers of the "first" child Buffer of the TwinBuffer, each with its associated `subBufferProperties` (using `srSDFEdgeBuffers` map will help).
			* Call the generateSubBuffers() function to generate the SubBuffers associated to the "second" child Buffer of the TwinBuffer (which will overwrite info from the "first" child Buffer in the `srSDFEdgeBuffers` map.
			* For each SubBuffer of the 2 childs of the TwinBuffer received as a parameter to the generateTwinSubBuffers function:
			    * Create a TwinBuffer whose two childs are the SubBuffers of the "first" and "second" corresponding to the same srSDFEdge (i.e. the same `subBufferProperties`). 
				* Associate this TwinBuffer to the srSDFEdge (i.e. the subBufferProperties) in the srSDFEdgeBuffers map.
 
* Check all usage of the Buffer in the codegenModelGenerator to make sure the TwinBuffer will not be an issue
    * From TwinBuffers stored in the `dagEdgeBuffers` Map
        * in generateCommunication(): Should be changed anyway to generate a communication that is not a "SharedMemoryCommunication".
		* in identifyMergedInputRange: dagEdgeBuffers is used to retrieve all pieces corresponding to a "NullBuffer" 
		  * Open question; Can a TwinBuffer be associated to a NullBuffer?
		      * Referring to the generateBuffer function: no, since the if statement of a TwinBuffer (the one leading to the exception line 880), is mutualy exclusive with the else leading to the creation of a NullBuffer.
			  * Nevertheless, a problem may arise if a non-"SharedMemoryCommunication" is created for a NullBuffer. But this is another problem, different from the twinBuffer. (a problem when creating communications...)
	* From TwinBuffers stored in the `srSDFEdgeBuffers` Map
        * TODO: Check the CodeGenHierarchicalModelGenerator class.(which receive the srSDFEdgeBuffers map as a param)
		* In generateBufferList(): I do not see any issue there. But check again just in case. 
		* In generateCallVariables(): I do not see any issue there. But check again just in case.
		* In generate FIFOCall(): same
		* In generateSpecialCall(): Same.

* In `CodegenAbstractPrinter()`:
    *  Add a `caseTwinBuffer`: Calls the caseBuffer on the child corresponding to the current "PrintedCoreBlock".

dmadronal, 2019-05-07:
A milestone on this support has been reached, as so far the buffer generation is quite similar to the one obtained with the MixedMerged memory distribution.

During this support, mainly the CodegenModelGenerator has been changed and two new classes have been included in the Codegen.xcore. Among the changes we can find:
 - Creation of a new class called TwinBuffer: this class stores the original buffer (original == first one added -- Equivalent to the shared memory buffer) and as many "twin" buffers as required (for explode/implode functionalities several twins are received)
- Creation of a new class called DistributedMemoryCommunication: This class aims to contain equivalent information than the SharedMemoryCommunication but has been created to ease the support from each platform
- During the whole execution of CodegenModelGenerator generate() function, the dagEdgeBuffers and srSDFEdgeBuffers have been checked to include in each case ONLY the buffer/subBuffer/Variable that was required
- This affects to addBuffer(), generateSpecialCall(), generateCallVariables() and generateBufferList() functions

leos313, 2019-05-07:
Before the end of the week, I will check the (eventually) compatibility/incompatibility with the Code-generation for Hardware on the FPGA. This is because the CodegenModelGenerator heavily affects the Codegen for the Hardware. In the TODO list, I have also the "Travis tests" to add in order to avoid this (now manual) operation.

dmadronal, 2019-05-07:
As long as you don't use the "DistributedOnly" option in the distribution field of Memory Allocation task your CodeGen should be safe. I also added a new print (actually not printing anything) in your Hardware CodeGeneration to avoid errors. At some point (if it makes sense in your case) you should implement that method to support that option.

As it was not supported before, I imagine that you are not using that option (so far) so I think you will not have problems.

dmadronal, 2019-05-16:
Update:

- Working for the MPPA, so I assume that the buffers are properly generated
- Supported also Clustered actors --> Hierarchical actors

dmadronal, 2019-05-22:
With the current status, I would say that the support for DistributedOnly memory allocation is missing:
- Support for the rest of the architectures
- Optimization in the buffer allocation and/or buffer exclusion --> When doing explode and implode, buffers associated to other PEs are also allocated in memory, e.g., if we do an explode from PE0 to the rest of PEs, PE1 will allocate a buffer corresponding to the transmission from PE0 to PE2. This buffer will never be used by PE1 and its allocation should be avoided to save memory space