preesm/preesm #146: Clustering not working properly --> execution segfault

dmadronal, 2019-05-13:
I have an issue when using the Clustering task on a PREESM workflow.

I'm working with the stabilization project and it is working fine (for 1 x86 PE) when using a workflow where all the network is flattened and no clustering is actually performed (CodegenDistribFlatMixedMerged.workflow). However, when using the Clustering task with the (CodegenDistribNoFlatMixedMerged.workflow) the application crashes when executing it.

I have seen that, for example, the offsets of the generated clustered functions are not considering the size of the variables, which can be one of the problem sources but I tried to change it manually and the segfault is still there.

I attach a project to replicate the issue. The sequence can be downloaded from [here](http://i.imgur.com/WMqwno7.mp4) and converted using this command:
`ffmpeg -i WMqwno7\ -\ Imgur.gif -vf scale=360:202 -c:v rawvideo -pix_fmt yuv420p JaguarAndCroco360x202-375f.yuv`
[org.ietr.preesm.stabilization_failing.tar.gz](https://github.com/preesm/preesm/files/3172071/org.ietr.preesm.stabilization_failing.tar.gz)

kdesnos, 2019-05-13:
Hi Daniel,

An internship is scheduled this summer to fix (or re-design) & improve the clustering task of Preesm.
Can it wait?

Karol

dmadronal, 2019-05-13:
Hi Karol,

Sure! It can wait on my side, don't worry =).

Dani

dmadronal, 2019-05-15:
Just more info:

I have tested the clustering on x86 with the sobel app (only 1byte variables) and it works, so I would say that the issue is related with the size of the token variables.