preesm/preesm #162: (Codegen) Compilation error when using specific hw configuration in the scenario

ClaudioRubattu, 2019-07-01:
The error is related to the sobel application of one of the provided tutorials (in attached the PREESM project).
[sobel_with _errors.zip](https://github.com/preesm/preesm/files/3345774/sobel_with._errors.zip)

When I consider the workflow "Codegen_Papify" and some specific scenarios (for instance "SOBEL_2PE_0A7_2A15_Papify"), I got the error in the attached txt file.
[error_on_sobel.txt](https://github.com/preesm/preesm/files/3345776/error_on_sobel.txt)

dmadronal, 2019-07-01:
Which PREESM version do you have?

antoine-morvan, 2019-07-01:
Bug occurs when the scheduler decides to map nothing on the "main operator" defined in the simulation tab. In such situation, the main operator ID value is set to -1.

commit 20f7f15b (2019-07-01): (Codegen) Fix #162
closed 2019-07-01 by antoine-morvan