preesm/preesm #178: Compile error when using g++ and several cores, cannot find shared variables

ggendro, 2019-07-18:
On version 3.11.0. When generating codegen, core0.c does not specify that its shared variables are "extern".
Then, if the architecture uses several cores and the compiler used is g++, the Makefile cannot compile, it cannot find the references to the shared variables.

The problem does not happen if compiler is gcc or if the application runs on a single core, only when both conditions are fulfilled.

Solutions could be:
1. To add "extern" keyword during code generation (generates a warning at compilation when using c compiler)
2. To be able to specify the extension of the generated files (c or cpp) and automatically add "extern" keyword when extension is cpp

antoine-morvan, 2019-07-22:
Could you attach a small project to reproduce ?

ggendro, 2019-07-22:
Here it is:
[org.ietr.preesm.reinforcement_learning.zip](https://github.com/preesm/preesm/files/3417713/org.ietr.preesm.reinforcement_learning.zip)

I simply took the RL Preesm app from [https://github.com/preesm/preesm-apps]( https://github.com/preesm/preesm-apps), changed the extension of C files from .c to .cpp and updated the cmake to make it search for cpp files.

antoine-morvan, 2019-07-22:
I see 2 issues in the project attached.

1. The CMakeFile.txt is missing the line `./codegen/*.c` in the file() section near the end. Thus the generated files (.c) are not compiled. This is OK when running 
2. The files in ./src/ are handled by the user. It is not Preesm responsibility to enforce the user .cpp files have the extern C {} sections properly declared.

While running the code initially, I also noticed that the codegen step does not delete .cpp files in the generated folder (./codegen in your example). That causes cmake to compile files that are out of date. The generated files should not be modified.

However SpiderCodegen do generate cpp files, and that could become an issue when playing with both in the same project. I can make sure the whole folder is cleared everytime the codegen task is called.

@kdesnos your thoughts?

closed 2019-09-13 by antoine-morvan