preesm/preesm #182: Execution Time of the 1-core Gantt generated is not the sum of the precharacterized actors

ClaudioRubattu, 2019-07-22:
The issue is verifiable considering the attached project (the Sobel tutorial in the repo). 
[sobel_gantt_issue.zip](https://github.com/preesm/preesm/files/3416820/sobel_gantt_issue.zip)

Running the workflow "Codegen_Papify" (*) with the scenarios "SOBEL_1PE_0A7_1A15_Papify_4customActivity" and "SOBEL_1PE_1A7_0A15_Papify_4customActivity", all actors are properly mapped into the only one present core (1 A15 or 1 A7 respectively). However, having precharacterized the timing of the actors, the sum of the weight should correspond to the final execution time, but 200 cycles are added with respect to the expected value.

For instance, regarding the first mentioned scenario, the timing is the following:
[image]
 
Considering the 4 repeats of the Sobel actor, the sum is 1035.
But in the Gantt appears 1235, as in the figure below.
[image]
   
I verified the same offset (200 cycles) in the second scenario. 

(*) : there is not PAPIFY task, and the monitored code generation is properly included into the Codegen task. This just to say this issue should not have any link with PAPIFY context.

antoine-morvan, 2019-07-22:
Split and Merge actors induce insertion of 1 explode and 1 implode actors in the SRDAG. Those actors have no timingn, thus default timing of 100 is used.

antoine-morvan, 2019-07-22:
@ClaudioRubattu I've seen the workflow you are using is pretty old and based on the old SDF model. I tried to run the new workflow tasks with your scenario, and the timings of explode/implode nodes are indexed on the memory copy speed.

antoine-morvan, 2019-07-22:
Using new workflow tasks (scheduling from PiMM), the explode/implode nodes have timings indexed on the memory speed: `timing = (fifo size / memory speed) + mem setup time`

closed 2019-08-21 by antoine-morvan