preesm/preesm #188: Monitoring an hierarchical actor with PAPIFY

leos313, 2019-07-23:
Is there a way of monitoring (measuring time) hierarchical actor of the PiSDF?
The PAPIFY tab gives me the possibility of selecting just the top of a hierarchical actor but it does actually nothing.

dmadronal, 2019-07-23:
Hi,

You can monitor all the actors contained in the hierarchical actor but not the complete hierarchical actor as a whole.

By monitoring all the actors inside a hierarchical actors (or only the first and the last ones) you can compute the timing of the whole hierarchical actor manually or using PAPIFY-VIEWER.

leos313, 2019-07-23:
Nope: there is the overhead of writing the file in the middle. For example:

```
#ifdef _PREESM_PAPIFY_MONITOR
event_stop_papify_timing(&papify_actions_nelderMeadCore_0_Centroid_0,0/*PE_id*/); // nelderMeadCore_0_Centroid_0
event_write_file(&papify_actions_nelderMeadCore_0_Centroid_0,0/*PE_id*/); // nelderMeadCore_0_Centroid_0
#endif
```

dmadronal, 2019-07-23:
If you want to not consider this overhead, then, you should monitor everything inside the hierarchical actor and add up all the partial results.

leos313, 2019-07-23:
nope: many of them are executed in parallel in different PE. 
What I can do, and I do, is to insert manually the instruction where is necessary. The question was exactly if there is an automated mechanism that I wasn't using properly!

dmadronal, 2019-07-23:
Hi again,

As mentioned in the first comment, the monitoring of a hierarchical actor as a whole is not supported right now. Since PAPIFY instructions are inserted during the code generation workflow task, at this point, the algorithm network is already flattened and the monitoring is performed in an "atomic actor" way.

In the scenario tab, the hierarchy is shown so as to ease and speed up the monitoring configuration.

Best,

Dani

leos313, 2019-07-23:
yeah! If the question has no sense and the feature will not be planned I think we can close the issue, right?
Thank you for answering!
Best 
Leos

closed 2019-08-21 by antoine-morvan