preesm/preesm #190: Error when generating .diagram if delay is linked to data interface

ggendro, 2019-07-25:
On version 3.13.1. Preesm cannot execute auto layout of a graph when a data input/output interface is linked to a delay with persistence level set to None.

When trying to generate .diagram, raises the following error : index=-2, size=0.

Here is a small project to reproduce the error: 
[test_auto_layout_error.zip](https://github.com/preesm/preesm/files/3430133/test_auto_layout_error.zip)

Generating diagram of the test_auto_layout.pi file should raise the error.

ahonorat, 2019-07-25:
This issue is most probably related with issue #128 .

ahonorat, 2019-07-31:
Actually, it may be also linked with issues #74 and #83 
Also another bug related: when deleting an actor having an output edge with a delay on it, the delay is not deleted.

commit 4703e05b (2019-09-04): Fix #74, Fix #83, Fix #128, Fix #190
commit 6ea6007f (2019-09-18): Fix #74, Fix #83, Fix #128, Fix #190