preesm/preesm #205: Error when running workflow if initialization of delay is linked to data interface

Martayugui, 2019-08-21:
Initialization of a delay using a data interface, generates an error when trying to run the workflow.
The following message appers:

**"SEVERE: Interface interfaceInput does not exist, or is not connect in vertex fatherActor hierarchy"**

Here, there is an evidence to reproduce the error:

[image]

antoine-morvan, 2019-08-21:
Reading the log, it seems you are using deprecated workflow tasks PiMM2SDF (see https://preesm.github.io/docs/workflowtasksref/#static-pimm-to-ibsdf---deprecated)

Martayugui, 2019-08-21:
I took the workflow from a project of my colleagues, I just replaced the deprecated one by the updated version. It works properly.

Now, I would like to use asymetric configuration for delay setter/getter actor, would it be possible to implement it?. In order to avoid the current error, I have to use both of data input/output interfaces in the delay configuration.

orlyn44, 2019-08-21:
As of now, asymetric configurations are not supported for setter / getter. 
The only possibility for you is to use a "dummy" actor that will consume the tokens in a setter / void configuration or that will produce dummy tokens in a void / getter configuration.
Those actors can be as simple as empty function calls, but they have to be present for the code generation to work.

closed 2019-08-21 by antoine-morvan