preesm/preesm #207: Problem with Tutorial Introduction

pablosdr, 2019-08-21:
Hi,

I've just installed PREESM in my Ubuntu 14.04 machine. I downloaded the files of the Tutorial Introduction and tried to run it but I got these messages:

NOTICE: Starting to build CPN list
NOTICE: Building CPN list.
NOTICE: Adding CPN actors to CPN and IBN actors in CPN dominant list
NOTICE: Mapping
NOTICE: Mapping finished, now add communication tasks.
NOTICE: DAG fully mapped, now removes useless sync and check schedules.
NOTICE: removing 0 syncs
NOTICE: Workflow Step: 'PiSDF Export' (id 'pisdf-export'): Exporting algorithm graph
SEVERE: Missing workflow task parameters: the task 'PiSDF Export' (id 'pisdf-export') requires parameters (hierarchical)

Any clue on why is this happening?

antoine-morvan, 2019-08-21:
Recent update in Preesm added a parameter to the workflow task that exports PiSDF IR.

When the workflow executor encounters a task for which some parameters are not set, it throws a warning. By default also, it fails on warning.

2 workarounds:

1. Add the parameter in the PiSDF Export task of the workflow (actually just moving one task and saving the workflow will add the default values for all parameters);
2. Configure the workflow to not fail on warnings (see https://preesm.github.io/release-3.3.0/)

I will update the tuto files to avoid failing with default settings.

antoine-morvan, 2019-08-21:
Fixed with https://github.com/preesm/preesm.github.io/commit/5fae68db6633648587b81bac116703c4c000917d

closed 2019-08-21 by antoine-morvan

pablosdr, 2019-08-21:
I followed the first workaround and it worked, thank you!