preesm/preesm #210: SEVERE: !PiMM2SrDAG refers a deprecated task

leos313, 2019-08-26:
I am getting this error after a `git pull` and `git rebase`.

```
20:53:38 NOTICE: Starting workflow execution: /tutorialSummerSchoolFixedTile/Workflows/CodegenPapify.workflow.
20:53:38 NOTICE: Workflow Step: 'PiMM2SrDAG' (id 'org.ietr.preesm.experiment.pimm2srdag.StaticPiMM2SrDAGTask'): Transforming PiGraph to Single-Rate Directed Acyclic Graph (MapperDAG).
20:53:38 SEVERE: !PiMM2SrDAG refers a deprecated task (org.ietr.preesm.experiment.pimm2srdag.StaticPiMM2SrDAGTask)!
```

Is there a new workflow to be used? Can I find the new workflow within the tutorial?

ahonorat, 2019-08-27:
Use the task 'pisdf-srdag' instead.
Same for the scheduling, now it is: 'pisdf-mapper.list'. 
Tell me if it solves the problem!

leos313, 2019-08-27:
I have checked with Claudio also and it works. No problems.

closed 2019-08-27 by ahonorat

leos313, 2019-08-27:
still a warning:
`11:51:35 WARNING: !DAG Exporter refers a deprecated task (org.ietr.preesm.mapper.exporter.DAGExportTransform)!`.
The workflow used:

[NewCodegenPapify.workflow.zip](https://github.com/preesm/preesm/files/3545471/NewCodegenPapify.workflow.zip)

antoine-morvan, 2019-08-27:
Following https://github.com/preesm/preesm/issues/205 I added warnings when deprecated tasks are encountered. Since there has not been a new release yet, no news is present on the website to warn users. The log entry is turned into severe and workflow fails because the general workflow parameter "Error on warnings" is set to true (see https://preesm.github.io/release-3.3.0/).