preesm/preesm #211: pushing on the remote: doubt

leos313, 2019-08-27:
I need an advice:

after a `git pull` and `git rebase` I solved all the merge conflicts and re-done locally the tests with no errors and failure.

However, when I push on the remote I have the output is:

```
To https://github.com/preesm/preesm.git
 ! [rejected]        papifyHardwareSupport -> papifyHardwareSupport (non-fast-forward)
error: failed to push some refs to 'https://github.com/preesm/preesm.git'
hint: Updates were rejected because the tip of your current branch is behind
hint: its remote counterpart. Integrate the remote changes (e.g.
hint: 'git pull ...') before pushing again.
```

is it enough to force the push on the remote with `git push -f ...`?

leos313, 2019-08-27:
maybe `git push --force-with-lease` can be a better solution?

ahonorat, 2019-08-27:
If you have been the only person to work on your branch (and always on the same computer), ```git push --force``` is ok.
@antoine-morvan probably has a better answer though.

leos313, 2019-08-27:
to be honest, I have been reading about that on the web and, finally, I forced a push and Travis seems not complaining about it. I think it is the fastest solution. Thank you Alex

closed 2019-08-28 by antoine-morvan