preesm/preesm #212: Issue in Tutorial 2

pablosdr, 2019-08-27:
Hi,

I am following the instructions for the tutorial about parallelizing an application on a multicore CPU. I am using an Ubuntu 16.04 machine.

- I downloaded the tutorial files, imported the project files in PREESM and placed the YUV and font files in $PROJECT/dat folder
- I successfully generated the C code by running the PREESM workflow
- I compiled the application by running the script $PROJECT/Code/CMakeGCC.sh
- When I try to run a.out in both $PROJECT/Code/bin/make/CMakeFiles/3.5.1/CompilerIdC and $PROJECT/Code/bin/make/CMakeFiles/3.5.1/CompilerIdCXX (I don't know which is the right one), nothing happens, even after granting execution permits.

Did I miss something?

Thank you very much for your response,

Pablo

ahonorat, 2019-08-27:
Just to be sure, have you executed the command "make" while in $PROJECT/Code/bin/make ?
Apparently this command is not in the script as it should.

After that, the executable to run is named "sobel", located in $PROJECT/Code/bin/make/Release.

pablosdr, 2019-08-27:
You are absolutely right, thanks!

Anyway, the generated sobel file throws a "segmentation fault (core dumped)" error when executed. What did I do wrong?

ahonorat, 2019-08-27:
Ok, I have updated the tutorial to fix the script and to ask the read to check the font path.
See https://github.com/preesm/preesm.github.io/pull/12 for the modifications.

closed 2019-08-27 by ahonorat