preesm/preesm #221: Issue with instrumented code in tutorial 07

pablosdr, 2019-08-30:
Hi,

I am following the tutorial 07 in order to measure the execution time of each actor in the Sobel algorithm.

The results found in analysis.cvs after running the instrumented code are:

top_sobel/Read_YUV | 23820
**top_sobel/Sobel | 12878**
top_sobel/Merge | 4503
top_sobel/display | 299335
top_sobel/Split | 4604
TIMINGS WEIGHTED SUM | 435286

I have a Intel® Xeon(R) CPU E5-1620 v3 @ 3.50GHz × 8 processor, and these results seems a bit weird to me since the fps achieved for the rest of the tests have been always above the numbers given by the tutorial (~2000 fps), and in this case an average time of 282 µs for the Sobel actor is given in the text (Intel i7 at 2.40GHz).

According to these results, the generated Gantt diagram shows very little parallelization capabilities:

[image]
 
Did I do something wrong when instrumenting my code?

Thanks!

Pablo

kdesnos, 2019-08-30:
Hi Pablo,

These results may simply indicate that your processor is more powerfull than what we expected when setting up the tutorial a few years ago :)
Basically, the yuv video provided within the tutorial has become too "small" in resolutions for modern processors. The sobel filter of such a small video is already very efficiently executed on a single core of your processor, and thus, parallelizing it will bring you virtually no benefits since the parallelization overhead (communications/ synchronization, ...) will be larger than the benefits of parallel computations :s.

My advice is to select a larger video (HD or even 4k) to better benefit from the compuation parallelization. To have even better result, in the code of the sobel filter, the following line:
```output[j*width + i] = (abs(gx) + abs(gy))/8;```
can be replaced with this:
```output[j*width + i] = sqrt(gx*gx + gy*gy);```

The tutorials will be updated in fall to include all these changes.

Best,

Karol

ahonorat, 2019-08-30:
Hello,

by the way, timings are given in nanosecond in the analysis.csv file. I just ran the Instrumented version and I found similar results as you. 
Also, when running the executable with instrumentation, the fps computed internally is usually wrong. This is because instrumentation modifies the number of firings of each actor.

Alex H.

pablosdr, 2019-08-30:
Thank you very much Karol and @ahonorat for your explanation :D Everything is more clear to me now!

closed 2019-09-13 by antoine-morvan